probonopd/WirelessPrinting #4: Implement Printer Monitor

probonopd, 2017-03-05:
To implement Printer Monitor, we need to get data from the printer all the time. UM3 does this by polling the device for JSON files all the time. Is there a better (and more resource-saving) way to do this on the ESP8266, e.g., using MQTT?

[image]

probonopd, 2017-03-06:
Switched the printing away from outputting a HTTP response at the same time. So we could request the status from time to time via HTTP (need to check if this can delay the printing output).

So the question is, what is the best logic to calculate the printing time. Can we trust on Cura putting in
```
;LAYER_COUNT:52
;LAYER:0
```
all the time and use that for the percentage calculation, and apply that to the time already printed?

probonopd, 2017-03-07:
Possibly comparing the

```
;TIME:15495
```

calculated by Cura with the seconds actually spent since printing began?

probonopd, 2017-03-12:
As per @me-no-dev, need to use https://github.com/me-no-dev/ESPAsyncWebServer for this.

probonopd, 2017-03-25:
Blocked by https://github.com/probonopd/WirelessPrinting/issues/12

probonopd, 2017-07-23:
We need to improve the temperature parsing. Here is a test sketch that can be used for this. Try with as many real-life lines from the printer as possible.

```
void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // prints title with ending response break
  Serial.println("ASCII Table ~ Character Map");
}

String parseTemp(String response, String whichTemp, bool getTarget=false) {
  String temperature;
  int Tpos = response.indexOf(whichTemp + ":");
  if (Tpos > -1) { // This response contains a temperature
    int slashpos = response.indexOf(" /", Tpos);
    int spacepos = response.indexOf(" ", slashpos + 1);
    //if match mask T:xxx.xx /xxx.xx
    if (spacepos - Tpos < 17) {
      if(! getTarget){
        temperature = response.substring(Tpos+whichTemp.length()+1, slashpos);
      } else {
        temperature = response.substring(slashpos+2, spacepos);
      }
    } else {
      temperature = "0.0";
    }
  }
  return(temperature);
}

// Variables for rinter status reporting
String temperature_actual;
String temperature_target;
String temperature_tool0_actual;
String temperature_tool0_target;
String temperature_bed_actual;
String temperature_bed_target;

void parseTemperatures(String response){
  temperature_actual = parseTemp(response, "T");
  temperature_target = parseTemp(response, "T", true);
  temperature_tool0_actual = parseTemp(response, "T0");
  temperature_tool0_target = parseTemp(response, "T0", true);
  temperature_bed_actual = parseTemp(response, "B");
  temperature_bed_target = parseTemp(response, "B", true);
}

void loop() {

  // String response = "ok T:32.8 /0.0 B:31.8 /0.0 T0:32.8 /0.0 @:0 B@:0";
  String response = "ok T:35.2 /0.0 B:31.7 /0.0 T0:35.2 /0.0 @:0 B@:0";

  parseTemperatures(response);

  Serial.print("temperature_actual: ");
  Serial.println(temperature_actual);
  Serial.print("temperature_target: ");
  Serial.println(temperature_target);

  Serial.print("temperature_tool0_actual: ");
  Serial.println(temperature_tool0_actual);
  Serial.print("temperature_tool0_target: ");
  Serial.println(temperature_tool0_target);

  Serial.print("temperature_bed_actual: ");
  Serial.println(temperature_bed_actual);
  Serial.print("temperature_bed_target: ");
  Serial.println(temperature_bed_target);

  Serial.println();
  
  delay(2000);
}
```

probonopd, 2017-08-12:
Currently it does not work when the printer reports no T0 but only T:

```
ok T:36.33 /0 @:0
ok T:36.19 /0 @:0
```

probonopd, 2017-09-23:
> We need to improve the temperature parsing.

Indeed:
https://github.com/fieldOfView/OctoPrintPlugin/issues/59

`/api/printer` currently gets wrong values when used with Original Prusa i3 MK2S.

```
{
  "temperature": {
    "tool0": {
      "actual": 60.16 E:0 B:38.7,
      "target": :60.16,
      "offset": 0
    },
    "bed": {
      "actual": 38.7,
      "target": :60.16,
      "offset": 0
    }
  },
  "sd": {
    "ready": 1
  },
  "state": {
    "text": "Operational",
    "flags": {
      "operational": true,
      "paused": false,
      "printing": 1,
      "sdReady": 1,
      "error": false,
      "ready": true,
      "closedOrError": false
    }
  }
}
```

GMagician, 2019-03-02:
I think this may be closed

probonopd, 2019-03-02:
Indeed. Thanks @GMagician :+1:

commit 7a42e2e1 (2017-08-12): Implement Printer Monitor, closes #4
closed 2019-03-02 by probonopd