probonopd/WirelessPrinting #5: Implement "Abort Print" function

probonopd, 2017-03-05:
Implement "Abort Print" function.

How do I do this, given that during printing a HTTP connection is open and I would need to get this to the printer despite the fact the esp8266/Arduino HTTP library only allows for one connection at a time?

probonopd, 2017-03-25:
Blocked by https://github.com/probonopd/WirelessPrinting/issues/12

probonopd, 2017-07-24:
Blocked by https://github.com/me-no-dev/ESPAsyncWebServer/issues/195

probonopd, 2017-08-12:
There is also 

> M410 - Quickstop
> Stop all steppers instantly
> 
> Stop all steppers instantly. Since there will be no deceleration, steppers are expected to be out of position after this command.

http://marlinfw.org/docs/gcode/M410.html

probonopd, 2019-03-01:
Actually this seems to be **working**, at least from Cura:

https://github.com/probonopd/WirelessPrinting/blob/427d8d4d9411a73ebf3416fc4d4ece6d374b3820/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino#L743-L744

It would be nicer, though, if we could somehow extract the end gcode and send that, or at the very least move the nozzle up a few centimeters before the M112.

GMagician, 2019-03-01:
The real problem is...every slicer write the same info to let us be able to do that?
Otherwise we have to write a per slicer handling (and also scan and interpreter gcode to find these info)

ipepe, 2019-03-02:
I'm not familiar but You could check into OctoPrint what actually happens there. It would be shame to restart printer after aborting?

chepo92, 2019-03-03:
What about an user configurable endgcode?

probonopd, 2019-03-04:
> configurable

Let's try to avoid configurability where we can. Whatever can be configured can be configured wrong.

ipepe, 2019-03-23:
I think that current abort print is a bit too excessive. Can we just stop sending GCODE commands to printer? Power cycling whole printer seems a bit excessive.

probonopd, 2019-03-23:
Imagine you are not at home and you see via the camera that something terribly bad is going on. I guess you want to reset the printer in those cases. Also, we are now sending more than one line of GCODE, so that simply sending no more might still result in activity for a while which may be undesirable.

ipepe, 2019-03-23:
I'm never leaving printer alone because I don't trust it 🤷‍♂️ Well, could we have at least both? Graceful stop, and nuke from the orbit stop?

ipepe, 2019-03-23:
We could make it a "defined" variable an I could use M410 instead of M112

probonopd, 2019-03-24:
What advantage does M410 give you over M112?

ipepe, 2019-03-24:
I assume that I don't have to power cycle my printer after M410?

probonopd, 2019-03-24:
I don't know but what is the problem with power cycling? If it increases safety...

ipepe, 2019-03-24:
Because it takes so much longer. When You abort a print on OctoPrint it doesn't do this. If I'm aborting print, it's because I changed my mind, or I can see that it will fail, or I chose wrong gcode (and noticed during heating that I want another one). I'm not having emergency 5 times a day, I'm just aborting a print. 

Emergency Stop != Abort Print

probonopd, 2019-03-24:
> Emergency Stop != Abort Print

Agree, let's have 2 buttons. Pull request welcome.

ipepe, 2019-03-24:
What about abort print from Cura?

probonopd, 2019-03-24:
Isn't Cura sending C-code? I have not checked how it works.

ipepe, 2019-03-24:
https://github.com/probonopd/WirelessPrinting/blob/master/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino#L307

positron96, 2019-11-28:
Any news on this? Every time I abort a print, I get very upset about having to power cycle printer - ESP gets power cycled as well, Cura looses connection, a somewhat frustrating experience. 
Or is this project abandoned?

probonopd, 2019-11-29:
Pull requests welcome.

GMagician, 2019-11-29:
AFAIK M112/M410 may not be executed immediately. It depends on how firmware is compiled (EMERGENCY_PARSER). In worst case it will need to purge all queue commands
 before they can do their job.
With EMERGENCY_PARSER best will be M410 (abort all commands) and a custom gcode (on SD or hardcoded)

positron96, 2019-11-30:
For me, every other option would be better than current situation since it moves the printer into inoperable state until power cycled. 
Actually just stopping sending further codes and letting the printer finish its queue seems acceptable. Above @probonopd described a use-case with monitoring the print via webcam remotely and needing to shut the machine immediately. I would argue that if printer has been failing for some time (you don't watch every second of printing via webcam, do you?), a couple of extra seconds of printing would not matter much.

probonopd, 2019-11-30:
What does matter though is that the print will be stopped with certainty, and the hotend and heatbeds will be switched off with certainty.

GMagician, 2019-12-04:
Both M112 and M410 turn off heaters and steppers (Position is not valid after M410 since it abruptly stop moving motors, not sure about m112)