probonopd/WirelessPrinting #7: When selecting printer, should bring up network dialog

probonopd, 2017-03-05:
When selecting an Ultimaker 3 as the printer, then the network selection dialog below comes up automatically. How can this be done for other printers?

[image]

probonopd, 2017-03-05:
It's probably due to the

```
{
    "id": "ultimaker3",
    "version": 2,
    "name": "Ultimaker 3",
    "inherits": "ultimaker",
    "metadata": {
        "author": "Ultimaker",
        "manufacturer": "Ultimaker",
        (...)
        "first_start_actions": [ "DiscoverUM3Action" ],
        "supported_actions": [ "DiscoverUM3Action" ],
        "supports_usb_connection": false
    },
(...)
```

probonopd, 2017-03-05:
Indeed, adding

```
        "first_start_actions": [ "DiscoverWirelessPrintAction" ],
        "supported_actions": [ "DiscoverWirelessPrintAction" ]
```

to the `metadata` of a machine definition file does the trick. Which leads us to the question, is it really clever to need this in each and every machine definition. WirelessPrinting can be used with every printer that accepts G-Code, so it would be good if Cura had an option to hook such plugins into every printer using a different mechanism that doesn't involve spamming all machine definition files.

fieldOfView, 2017-03-06:
See how the OctoPrintPlugin attaches itself automatically to all printers supporting a usb connection:
https://github.com/fieldOfView/OctoPrintPlugin/blob/2.4/DiscoverOctoPrintAction.py#L64

Note that octoprintplugin decides not to pop up the interface directly when adding a printer, but instead registers itself as an available plugin. If you really want the interface when adding the printer, use addFirstStartAction instead of addSupportedAction.

probonopd, 2017-03-06:
Thanks @fieldOfView. Yes, why would I _not_ want the interface when adding the printer?

closed 2017-03-06 by probonopd

fieldOfView, 2017-03-07:
I know *you* want the interface, but say this gets included in Cura by default. Now everyone gets the interface, on all printers, regardless if they have your wireless hardware.

commit 4c78852d (2017-03-06): addFirstStartAction, should close #7. Thanks @fieldOfView