probonopd/WirelessPrinting #8: Preheat extruder when starting upload?

probonopd, 2017-03-05:
Should we preheat the extruder, e.g., to 200 degrees C, when starting an upload, so that the heating already takes place while the code is still being uploaded?

What do people think?

probonopd, 2017-03-07:
This may be a safety issue. Can we ensure that the pre-heated temperature stays for e.g., 3 minutes max and if no further commands (e.g., new temperature setting) arrive, then it is cooled down again? (This is the standard behavior of the "Preheat PLA" feature of my Marlin printer firmware when used standalone).

probonopd, 2017-07-24:
Cura has a "pre-heat" button for the build plate at least. Using it is blocked by me-no-dev/ESPAsyncWebServer#195.

A13xund3r, 2019-01-29:
No, extruder heats up quite fast, but bed yes, depend on printer config and material it might be 5-15min for bed, similar to upload time, if one need extruder ready it could be preheated to 150, no more, some materials deteriorate if kept in hot nozzle too long.

GMagician, 2019-03-02:
In my optinion it's better to handle "cura" prehead request than add a duplicated function here

probonopd, 2019-03-02:
I think that https://github.com/probonopd/WirelessPrinting/issues/63 should take care of Cura preheat automatically.

This here is for all kinds of GCODE, not just Cura.

GMagician, 2019-03-02:
Then it's better to create a good webpage and put there commands

probonopd, 2019-03-02:
I love "automagic" where the user doesn't need to think of those things or do anything, where "things just work".

probonopd, 2020-03-17:
Closing for now since this can be done in the slicer, feel free if you think this should be reopened.

closed 2020-03-17 by probonopd