probonopd/WirelessPrinting #9: Stops printing after ca. 26 minutes

probonopd, 2017-03-05:
Had it stop printing after ca. 26 minutes. Is this because the HTTP request times out at some point? Would it be more stable if we did not send the printer <-> ESP8266 communication as a HTTP response while printing?

Can this be reproduced?

probonopd, 2017-03-05:
Let's see whether https://github.com/probonopd/WirelessPrinting/commit/cfde7b92c8cc7f5e229962216df229dc94045ddc addresses this.

closed 2017-03-05 by probonopd

dentex, 2019-07-23:
Hello.
I'm having the same problem, so I'm writing here instead of opening a new issue.

I'm using a generic ESP32 board (probably known as "development board"; I'm not sure. It's just a Chinese board I got from Aliexpress 
https://s.click.aliexpress.com/e/cYl5MYek)

The print seems to stop after about 10~20 minutes.

Other things I found are also that:
- no progress is shown on the printer LCD
and
- the pre-built version doesn't seem to work on my ESP32. I built it myself with the Arduino IDE manually adding the various libraries needed.