probonopd/WirelessPrinting #11: Basic setup

micktol, 2017-03-08:
I have uploaded the sketch to a nodemcu and can see the wifi is connecting, but when I try to connect from firefox there is no response. Should I be able to connect?
What is the default speed for the serial port? 
What value of capacitor do you use?

probonopd, 2017-03-08:
The default speed in the sketch is 115.200 bits/s but you can change it in the sketch in the `Serial.begin(115200);` line.

The exact value of the capacitor shouldn't matter much, I used 100 uF.

probonopd, 2017-03-25:
@micktol did you get it to work? Most SD shields already include the required capacitor.

micktol, 2017-03-25:
Yes, I had it working with and without the capacitor, I tried 3 prints but each time it stopped printing before the print was finished!

probonopd, 2017-03-25:
Can you point me to the file you tried to print? I have printed tens of different models so far, all without issues. How long is the calculated print time (in Cura) for the failed models?

micktol, 2017-03-25:
I was not printing from cura.
I sliced with simplfiy3d and then uploaded via web page.

probonopd, 2017-03-25:
In this case, can you point me to the file you tried to print? How long is the calculated print time (in simplfiy3d) for the failed models?

micktol, 2017-03-26:
Attached is an example, the slicer estimates 13 minutes 
[tube1.gcode.zip](https://github.com/probonopd/WirelessPrinting/files/870657/tube1.gcode.zip)

probonopd, 2017-03-26:
@micktol that is very strange, I have definitely printed successful prints that were much, much longer (2 hours and more). Currently I have no clue why this is. Could it be that your firmware responds with something else than "ok"? WirelessPrinting only sends the next line if the printer responded with "ok" before.

micktol, 2017-03-26:
I am using repetier firmware as I have a radds board.

probonopd, 2017-03-26:
It might have to do with that. Do you have a means to observe the data that is sent by the printer before the print stalls?

probonopd, 2017-04-07:
I observed the same.

```
echo:Unknown command: ""
ok
echo:Unknown command: ""
ok
echo:Unknown command: ""
ok
echo:Unknown command: ""
ok
echo:Unknown command: ""
ok
```

Turned out that my SD card needed to be reformatted.
Working again now.

probonopd, 2017-07-24:
@micktol does the problem persist when you use WirelessPrintAsync? It is rewritten almost from scratch using other libraries, and seems to work much better for me.

probonopd, 2018-04-09:
Can this be closed?

probonopd, 2019-02-28:
Apparently yes. Feel free to comment if this should be reopened.

closed 2019-02-28 by probonopd