probonopd/WirelessPrinting #18: Implement api/settings?apikey= for OctoPrintPlugin

probonopd, 2017-08-11:
OctoPrintPlugin for Cura now checks `api/settings?apikey=` which is currently unimplemented.

https://github.com/Ultimaker/Cura/issues/2257#issuecomment-321920065

probonopd, 2017-08-11:
Need to find out what is the minimal subset of a response that we can get away with. Here is the original from a real OctoPrint instance:

```
 "api": {
    "allowCrossOrigin": false, 
    "enabled": true, 
    "key": "__ 32 characters uppercase hex string __"
  }, 
  "appearance": {
    "color": "default", 
    "colorTransparent": false, 
    "defaultLanguage": "_default", 
    "name": ""
  }, 
  "feature": {
    "alwaysSendChecksum": false, 
    "externalHeatupDetection": true, 
    "gcodeViewer": true, 
    "ignoreIdenticalResends": false, 
    "keyboardControl": true, 
    "pollWatched": false, 
    "repetierTargetTemp": false, 
    "sdAlwaysAvailable": false, 
    "sdSupport": true, 
    "swallowOkAfterResend": true, 
    "temperatureGraph": true, 
    "waitForStart": false
  }, 
  "folder": {
    "logs": "/home/pi/.octoprint/logs", 
    "timelapse": "/home/pi/.octoprint/timelapse", 
    "timelapseTmp": "/home/pi/.octoprint/timelapse/tmp", 
    "uploads": "/home/pi/.octoprint/uploads", 
    "watched": "/home/pi/.octoprint/watched"
  }, 
  "plugins": {
    "cura": {
      "cura_engine": "/usr/local/bin/cura_engine", 
      "debug_logging": false, 
      "default_profile": null
    }, 
    "discovery": {
      "httpPassword": null, 
      "httpUsername": null, 
      "model": {
        "description": null, 
        "name": null, 
        "number": null, 
        "serial": null, 
        "url": null, 
        "vendor": null, 
        "vendorUrl": null
      }, 
      "pathPrefix": null, 
      "publicHost": null, 
      "publicPort": 80, 
      "upnpUuid": "01ee6300-ee9e-4ae0-a0f4-37f5787774ad", 
      "zeroConf": []
    }, 
    "pluginmanager": {
      "dependency_links": false, 
      "hidden": [], 
      "pip": null, 
      "pip_args": null, 
      "repository": "http://plugins.octoprint.org/plugins.json", 
      "repository_ttl": 1440
    }
  }, 
  "printer": {
    "defaultExtrusionLength": 5
  }, 
  "scripts": {
    "gcode": {
      "afterPrintCancelled": "; disable motors\nM84\n\n;disable all heaters\n{% snippet 'disable_hotends' %}\n{% snippet 'disable_bed' %}\n;disable fan\nM106 S0", 
      "snippets/disable_bed": "{% if printer_profile.heatedBed %}M140 S0\n{% endif %}", 
      "snippets/disable_hotends": "{% for tool in range(printer_profile.extruder.count) %}M104 T{{ tool }} S0\n{% endfor %}"
    }
  }, 
  "serial": {
    "additionalPorts": [], 
    "autoconnect": true, 
    "baudrate": 115200, 
    "baudrateOptions": [
      115200, 
      250000, 
      230400, 
      57600, 
      38400, 
      19200, 
      9600
    ], 
    "checksumRequiringCommands": [
      "M110"
    ], 
    "disconnectOnErrors": true, 
    "helloCommand": "M110 N0", 
    "ignoreErrorsFromFirmware": false, 
    "log": false, 
    "longRunningCommands": [
      "G4", 
      "G28", 
      "G29", 
      "G30", 
      "G32", 
      "M400", 
      "M226"
    ], 
    "port": "/dev/ttyUSB0", 
    "portOptions": [], 
    "timeoutCommunication": 30.0, 
    "timeoutConnection": 10.0, 
    "timeoutDetection": 0.5, 
    "timeoutSdStatus": 1.0, 
    "timeoutTemperature": 5.0, 
    "triggerOkForM29": true
  }, 
  "server": {
    "commands": {
      "serverRestartCommand": "sudo service octoprint restart", 
      "systemRestartCommand": "sudo shutdown -r now", 
      "systemShutdownCommand": null
    }, 
    "diskspace": {
      "critical": 209715200, 
      "warning": 524288000
    }
  }, 
  "system": {
    "actions": [
      {
        "action": "shutdown", 
        "async": true, 
        "command": "sudo shutdown -h now", 
        "confirm": "You are about to shutdown the system.", 
        "ignore": true, 
        "name": "Shutdown"
      }, 
      {
        "action": "reboot", 
        "async": true, 
        "command": "sudo shutdown -r now", 
        "confirm": "You are about to reboot the system", 
        "ignore": true, 
        "name": "Reboot"
      }, 
      {
        "action": "restart", 
        "async": true, 
        "command": "sudo service octoprint restart", 
        "confirm": "You are about to restart OctoPrint", 
        "ignore": true, 
        "name": "Restart OctoPrint"
      }
    ], 
    "events": null
  }, 
  "temperature": {
    "cutoff": 30, 
    "profiles": [
      {
        "bed": 100, 
        "extruder": 210, 
        "name": "ABS"
      }, 
      {
        "bed": 60, 
        "extruder": 180, 
        "name": "PLA"
      }
    ]
  }, 
  "terminalFilters": [
    {
      "name": "Suppress M105 requests/responses", 
      "regex": "(Send: M105)|(Recv: ok (B|T\\d*):)"
    }, 
    {
      "name": "Suppress M27 requests/responses", 
      "regex": "(Send: M27)|(Recv: SD printing byte)"
    }
  ], 
  "webcam": {
    "bitrate": "5000k", 
    "ffmpegPath": "/usr/bin/avconv", 
    "ffmpegThreads": 1, 
    "flipH": false, 
    "flipV": false, 
    "rotate90": false, 
    "snapshotUrl": "http://127.0.0.1:8080/?action=snapshot", 
    "streamUrl": "/webcam/?action=stream", 
    "watermark": true
  }

```

fieldOfView, 2017-08-11:
It doesn't have to return anything, an http return code of 200 suffices.

If you want the webcam stream to work, you need to return that section (specifically streamUrl, flipH, flipV and rotate90).

probonopd, 2017-08-11:
Thank you very much @fieldOfView 👍 that helps, since there is not _that_ much space on a 3 EUR microcontroller running WLAN and a subset of the OctoPrint protocol ;-)

probonopd, 2017-08-11:
Well, now getting

```
Version: 2.7.0-BETA
Platform: Linux-4.4.76-1-default-x86_64-with-SuSE-42.3-x86_64
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "/tmp/.mount_HxmBnT/usr/bin/plugins/plugins/OctoPrintPlugin/DiscoverOctoPrintAction.py", line 246, in _onRequestFinished
    json_data = json.loads(bytes(reply.readAll()).decode("utf-8"))
  File "/build/env/2.7/inst/lib/python3.5/json/__init__.py", line 319, in loads
  File "/build/env/2.7/inst/lib/python3.5/json/decoder.py", line 339, in decode
  File "/build/env/2.7/inst/lib/python3.5/json/decoder.py", line 357, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

-------------------------
Thread 0x00007f4eccdeb700 (most recent call first):
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Backend/Backend.py", line 155 in _storeStderrToLogThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ecd5ec700 (most recent call first):
  File "/build/2.7/build/inst/lib/python3/dist-packages/UM/Backend/Backend.py", line 147 in _storeOutputToLogThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ece5ee700 (most recent call first):
  File "/tmp/.mount_HxmBnT/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4eceffd700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ecf7fe700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ecffff700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ef4db2700 (most recent call first):
  File "/tmp/.mount_HxmBnT/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 58 in _updateThread
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ef55b3700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/2.7/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f4ef5db4700 (most recent call first):
  File "/build/env/2.7/inst/lib/python3.5/threading.py", line 297 in wait
[... 56 lines cut ...]
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
```

Sigh.

It does return 200 with an empty body:

```
inux@linux:~> curl -v http://192.168.0.23/api/settings?apikey=...
* Hostname was NOT found in DNS cache
*   Trying 192.168.0.23...
* Connected to 192.168.0.23 (192.168.0.23) port 80 (#0)
> GET /api/settings?apikey=83C3E33359A04446A177B6BDD722BD03 HTTP/1.1
> User-Agent: curl/7.37.0
> Host: 192.168.0.23
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: 0
< Content-Type: text/plain
< Connection: close
< Accept-Ranges: none
< 
* Closing connection 0
```

probonopd, 2017-08-11:
Possibly we need an empty JSON string, `{}`.

closed 2017-08-11 by probonopd

probonopd, 2017-08-11:
Confirmed working. Thanks @fieldOfView

commit c309a206 (2017-08-11): Return 200 on /api/settings, closes #18
commit 6fd0195e (2017-08-11): Return empty JSON for /api/settings in an attempt to close #18