probonopd/WirelessPrinting #21: Show information about the print progress on LCD

probonopd, 2017-08-12:
http://www.reprap.org/wiki/Firmware_Capabilities_Protocol

> M531 filename - Set the filename being printed.
> M532 X<percent> L<curLayer> - Update current print state progress (X=0..100) and layer L. 
> 
> This allows a host to enable a special information mode on LCD display, so users see progress in percent, model being printed, layer being printed, and total layer count. (Hosts can also use M117 )

GMagician, 2019-03-01:
This should be addressed by version 2.0. Only thing missing is L. Cura gives info in gcode but I don't know if all slicers does this and if they does, if they uses same format

probonopd, 2019-03-03:
We are already calculating the percentage for `/api/job` (`completion`), now we "just" need to find a way to send it to the LCD from time to time.

GMagician, 2019-03-03:
M73 is still supported if printer report capability to handle it

probonopd, 2019-03-03:
Only the last one works on the Prusa:

```
>M531 filename
<Invalid M code.#wait more#
<ok#ok#

>M532 75 20
<Invalid M code.#wait more#
<ok#ok#

>M73 75
<Invalid M code.#wait more#
<ok#ok#

>M117 75%
<ok#ok#
>M105
```

probonopd, 2019-03-03:
Second printer, Renkforce RF100 with OEM firmware:

```
>M115
<FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000#wait more#
<ok#ok#

>M531 filename
<ok#ok#

>M532 75 20
<ok#ok#

>M532 P75 20
<ok#ok#

>M73 75
<ok#ok#

>M117 Printing... 75%
<ok#ok#
```

We get "OK" here but no change on the LCD; only the last command does something visible.

GMagician, 2019-03-03:
all of that depends on firmware...if firwmware supports them then it is supposed it will do something on LCD

probonopd, 2019-03-03:
Yes, so let's use what works on _all_ (most) printers. Then we can improve later from thre.