probonopd/WirelessPrinting #25: enhancement request: Integration for autodrop3d

mmiscool, 2017-10-27:
Hello,

I have  been developing  system for a completely unmanned 3d printing. 
Currently I have been using a script running on a raspberry pi but I think that could be replaced with this project. 

Currently the pi downloads the gcode file from a http server and then prints it. After the item is printed and eject from the printer it notifys the server via a http request that it is competed and it repeats the cycle.

If it is possible to make the esp download a gcode from a server and store it to the sd card and then print from the sd card I think I could replace the pi in my set up.

Would be willing to contribute some cash and work on this with some one.

probonopd, 2017-10-27:
@mmiscool this sounds totally doable.

So, the ESP would ask your server for the next thing to be printed, download that gcode and store it, and after completion of that gcode inform the server that it is complete, and then start the next cycle.

Is autodrop3d already available in some form or is it still in the planning stage?

mmiscool, 2017-10-28:
@probonopd
I have it up and running and the source code is on git hub. I can give you a log in and password for the server if you want to take a look.

https://github.com/Autodrop3d

And yes. You summarised it very well there.

probonopd, 2017-10-28:
Currently WirelessPrinting announces itself to the network with ZeroConf so that the user can select a machine via the GUI built into e.g., the Cura OctoPrint plugin.

How does the device discovery work in your scenario? Does Autodrop3d discover printers on the network, or are the clients (the hardware attached to the printers) discovering the server, or does the user have to manually set this up?

mmiscool, 2017-10-28:
There is no discovery. When the printer contacts the server to request if a print job is available it sends the printers name, size and color.  On the server side this is checked against available approved print jobs. If a print job is available it returns the gcode for that print job otherwise it returns "no jobs available"

The gcode file it downloads has gcode comments for the first 10 lines and contains the job id, and some other information.

Things that would have tobe configured by the user would be the size of the print volume. material/color. printer name. url for server to fetch jobs from.

probonopd, 2017-10-29:
So on the ESP, we would need to have an interface for the user to enter these things? Isn't that a bit cumbersome, wouldn't it be easier if e.g., on the server, there was one central GUI for setting these things for a given client MAC address or serial number, and then have the clients automatically connect to the server without additional configuration? If the MAC address or serial number is known to the server, printing could start without the user having to configure anything on the client. This would also make the client code much easier since we would not have to edit and store these things on the ESP...

mmiscool, 2017-10-29:
The printers could be any where in the world and talk to the same server. The server could be hosted locally or in the cloud. That makes auto discovery and management a bit harder.

probonopd, 2017-10-29:
Interesting use case... so I'd say it's doable but it's more work than what I can do in my free time. If you want to work on it I'd support you, though.

mmiscool, 2017-11-01:
I started to look at trying to compile your project.

I used the latest arduino ide and the latest boards manager package for esp8266 and attempted to compile after fetching the dependencies list that needed to be added to the libraries folder. Unfortunately I was unable to compile.

I can post up the compiler output later when I get home. If you have the time would you mind zipping up a copy of your libraries folder so I can test and see if that dose the trick?

probonopd, 2017-11-02:
Please follow 1:1 the steps in the [`.travis.yml`](https://github.com/probonopd/WirelessPrinting/blob/master/.travis.yml) file.

probonopd, 2019-02-28:
Hi @mmiscool are you still on this?

probonopd, 2019-03-03:
Looks like development of autodrop3d has halted; feel free to comment if this is not the case and you would like this to be reopened.

closed 2019-03-03 by probonopd