probonopd/WirelessPrinting #26: Error While compiling

shaddow501, 2017-11-05:


shaddow501, 2017-11-05:
C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:78:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, pbuf*, long int)' to 'tcp_recv_fn {aka signed char (*)(void*, tcp_pcb*, pbuf*, signed char)}' [-fpermissive]

     tcp_recv(_pcb, &_s_recv);

                            ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:318:18: error:   initializing argument 2 of 'void tcp_recv(tcp_pcb*, tcp_recv_fn)' [-fpermissive]

 void             tcp_recv    (struct tcp_pcb *pcb, tcp_recv_fn recv) ICACHE_FLASH_ATTR;

                  ^

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:79:28: error: invalid conversion from 'long int (*)(void*, tcp_pcb*, uint16_t) {aka long int (*)(void*, tcp_pcb*, short unsigned int)}' to 'tcp_sent_fn {aka signed char (*)(void*, tcp_pcb*, short unsigned int)}' [-fpermissive]

     tcp_sent(_pcb, &_s_sent);

                            ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:319:18: error:   initializing argument 2 of 'void tcp_sent(tcp_pcb*, tcp_sent_fn)' [-fpermissive]

 void             tcp_sent    (struct tcp_pcb *pcb, tcp_sent_fn sent)ICACHE_FLASH_ATTR;

                  ^

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:80:28: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

     tcp_err(_pcb, &_s_error);

                            ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

 void             tcp_err     (struct tcp_pcb *pcb, tcp_err_fn err)ICACHE_FLASH_ATTR;

                  ^

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:81:31: error: invalid conversion from 'long int (*)(void*, tcp_pcb*)' to 'tcp_poll_fn {aka signed char (*)(void*, tcp_pcb*)}' [-fpermissive]

     tcp_poll(_pcb, &_s_poll, 1);

                               ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:320:18: error:   initializing argument 2 of 'void tcp_poll(tcp_pcb*, tcp_poll_fn, u8_t)' [-fpermissive]

 void             tcp_poll    (struct tcp_pcb *pcb, tcp_poll_fn poll, u8_t interval)ICACHE_FLASH_ATTR;

                  ^

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':

C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:130:25: error: invalid conversion from 'void (*)(void*, long int)' to 'tcp_err_fn {aka void (*)(void*, signed char)}' [-fpermissive]

   tcp_err(pcb, &_s_error);

                         ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:321:18: error:   initializing argument 2 of 'void tcp_err(tcp_pcb*, tcp_err_fn)' [-fpermissive]

[... 109 lines cut ...]

   tcp_accept(_pcb, &_s_accept);

                              ^

In file included from C:\Users\user\Documents\Arduino\libraries\ESPAsyncTCP-master\src\ESPAsyncTCP.cpp:27:0:

C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-nightly+20170222\libraries\SD
 Not used: D:\arduino-1.8.1\libraries\SD
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\user\Documents\Arduino\libraries\ArduinoJson
 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

probonopd, 2017-11-05:
Never tried to compile this on Windows. Compiles fine for me using the exact steps in https://github.com/probonopd/WirelessPrinting/blob/master/.travis.yml.

It is probably very picky about the exact versions being used.

probonopd, 2017-11-05:
If you just want to _use_ this, there is a pre-compiled version available at https://github.com/probonopd/WirelessPrinting/releases.

shaddow501, 2017-11-05:
Well downloaded all the git files 
  - git clone -o 56e7450 https://github.com/probonopd/ESPAsyncWebServer.git # Patched version until https://github.com/me-no/dev/ESPAsyncWebServer/pull/192 is merged
  - ( cd ESPAsyncWebServer ; git checkout 56e745 )
  - git clone https://github.com/me-no-dev/ESPAsyncTCP.git
  - git clone https://github.com/alanswx/ESPAsyncWiFiManager.git
  - git clone https://github.com/bblanchon/ArduinoJson.git

but still have the same errors

shaddow501, 2017-11-05:
Well I would like to find a way to add yours to the ESP3D, so it will also support CURA and support Repetier-host…

probonopd, 2017-11-05:
It already supports Cura... which ships with an OctoPrint plugin. We implement a subset of the OctoPrint API, just enough to work with Cura.

Not sure about Repetier-host though.

shaddow501, 2017-11-05:
I know it does support CURA but I want something to support both CURA & Repeter-Host…

probonopd, 2017-11-05:
Sorry for being ignorant, does Repetier-host implement the OctoPrint protocol?

shaddow501, 2017-11-05:
Lol I don’t know

closed 2018-04-09 by probonopd