probonopd/WirelessPrinting #27: Consider to support IPP 3D Extensions defined by ISTO Printer Working Group

KurtPfeifle, 2017-11-05:
You should consider to follow the 3D printing developments currently being worked on by the [ISTO Printer Working Group (PWG)](https://www.pwg.org/) with the goal of adding 3D related extensions to the Internet Printing Protocol (IPP).

- **Website:** https://www.pwg.org/3d/
- **Specs ("candidate standard"):** http://ftp.pwg.org/pub/pwg/candidates/cs-ipp3d10-20170210-5100.21.pdf
- **Spec ("best practices") for 3D JobTicket:** http://ftp.pwg.org/pub/pwg/informational/bp-smpjt3d10-20170818.pdf
- **Mailing List:** http://www.pwg.org/mailman/listinfo/3d-printing

<sub>BTW, the *`ippsample`* project already has support for 3D spec/standad testing already included. Just enable the *`transform3d`* binary to be built (which requires additional dependencies, f.e. Cura).</sub>

probonopd, 2017-11-05:
Thank you very muvh @KurtPfeifle - very interesting. Is this what Microsoft, too, uses in Windows to discover (some) 3D printers?

KurtPfeifle, 2017-11-05:
> *"Is this what Microsoft, too, uses in Windows to discover (some) 3D printers?"*

No idea about Microsoft. I suspect they currently rather follow [one of the Wi-Fi specs](https://www.wi-fi.org/discover-wi-fi/specifications) (maybe [WiFi-Direct](https://www.wi-fi.org/discover-wi-fi/wi-fi-direct)?) created by the [Wi-Fi Alliance](https://www.wi-fi.org/). This in turn however, "cloned" many ideas from various IPP specifications...