probonopd/WirelessPrinting #28: How to connect?

shaddow501, 2017-11-08:
I have programmed your code but I cant find out how to connect?? how to I find out the IP?

probonopd, 2017-11-08:
The Cura Octoprint plugin and PrusaControl should find the printer automatically using Zeroconf/Bonjour. No need to manually enter an IP address.

shaddow501, 2017-11-08:
So I do not need to configure  this?
[image]

probonopd, 2017-11-09:
No

probonopd, 2017-11-09:
I notice you are using Windows. Not sure about that. I have no Windows, am not interested in supporting it, and don't know how well Cura OctoPrint plugin works on Windows for detecting printers via Zeroconf. Maybe https://github.com/fieldOfView/OctoPrintPlugin knows.

weidmane, 2018-02-14:
https://learn.adafruit.com/bonjour-zeroconf-networking-for-windows-and-linux/overview

probonopd, 2018-02-16:
@weidmane thanks for the hint. Does it work for you if you follow that guide? As said above,

> I have no Windows, am not interested in supporting it

but if you can send a pull request that makes it work on Windows (and doesn't break it) I'm happy to include it. Thanks.

probonopd, 2018-04-09:
> how to I find out the IP?

The IP should be displayed on the display of your printer. Is it?

fieldOfView, 2019-02-26:
For the record, the Cura OctoPrint Connection plugin uses a python-based implementation of Bonjour/Zeroconf, so it does not need an additional install on Windows.

probonopd, 2019-02-28:
I can confirm that on Windows 10 the WirelessPrinting instance is not found by Cura 3.6 with the OctoPrint Plugin 3.5.4.

On Linux it works.

cpeuschel, 2020-03-16:
Couldnt reproduce that on Cura 4.5 with OctoPrint Plugin 3.5.12
[image]

probonopd, 2020-03-16:
Thank you very much for testing @cpeuschel. Closing for now, unless someone reports to reproduce the issue.

closed 2020-03-16 by probonopd