probonopd/WirelessPrinting #29: Support ESP32 (md5.h: No such file or directory)

probonopd, 2018-02-11:
Added beginnings of ESP32 support but am getting: `libraries/ESPAsyncWebServer/src/WebAuthentication.cpp:23:17: fatal error: md5.h: No such file or directory`

https://github.com/probonopd/WirelessPrinting/commit/668d14a5c19d57958d51a462f00124913917b214

probonopd, 2018-02-11:
https://github.com/tzapu/WiFiManager/tree/esp32 must be used, which states

> ESP32 must be patched with webserver library!!!!!! see https://github.com/tzapu/WiFiManager/issues/241

which points to

> I made a fork of esp32 with bbx10 webserver and captiveportal fix
> https://github.com/tablatronix/arduino-esp32

So do I actually need to use the https://github.com/tablatronix/arduino-esp32 core in order to be able to use WiFiManager on the ESP32?

probonopd, 2018-02-11:
Why ESP32?
* Dual core
* Bluetooth
* There are nice modules with integrated SD card slot available, e.g, the __TTGO T1 ESP-32__

Testato, 2018-02-13:
You can use the official esp32 core, but for use wifimanager you need:
- the bbx10 webserver (because esp32 core at moment do not include a webserver library)
- patch the dnsserver library (because the actual esp32 dnsserver have a bug)

We hope that in few time the espressifvteam solve this two problem officially

just-jason, 2019-01-05:
the ESP32 looking to be the way forward. Marlin are now supporting SDIO protocol, so the use of the ESP32+SD shield could be possible with very usable upload speeds.

https://github.com/MarlinFirmware/Marlin/issues/11641
https://github.com/MarlinFirmware/Marlin/pull/12782

https://yannik520.github.io/sdio.html

probonopd, 2019-01-05:
Does SDIO mean I can either have a SD card in the printer OR use SDIO? Then I don't see the advantage if I want to upload something to the printer's SD card...

just-jason, 2019-01-07:
At least on some boards it will mean that you can use the on board SD card,  but you are right, it is not a perfect solution, and not for all boards.

probonopd, 2019-03-03:
From this [log](https://api.travis-ci.org/v3/job/500964727/log.txt) it looks like `StorageFS.h` might need some IFDEFs to support ESP32 @GMagician?

```
In file included from sketch/FileWrapper.cpp:1:0:
FileWrapper.h:16:9: error: 'Dir' in namespace 'fs' does not name a type
     fs::Dir fsDir;
         ^
FileWrapper.h:43:5: error: invalid abstract return type 'FileWrapper'
     FileWrapper openNextFile();
     ^
sketch/FileWrapper.h:7:7: note:   because the following virtual functions are pure within 'FileWrapper':
 class FileWrapper : public Stream {
       ^
In file included from /home/travis/.arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/Arduino.h:153:0,
                 from /home/travis/.arduino15/packages/esp32/hardware/esp32/1.0.1/libraries/FS/src/FS.h:25,
                 from sketch/FileWrapper.h:4,
                 from sketch/FileWrapper.cpp:1:
/home/travis/.arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/Stream.h:51:18: note: 	virtual void Stream::flush()
     virtual void flush() = 0;
                  ^
```

probonopd, 2019-03-03:
Note: Since I was working on an older version (before the ArduinoJSON update), it may be easier to not use my branch but start from master and take over my changes
1. Edit .travis.yml
2. Move sketch directory into `examples` (https://github.com/adafruit/travis-ci-arduino needs it like that duh)

GMagician, 2019-03-04:
It seems stream has a flush method that is not implemented but in esp32 is required...just 2 seconds fix

GMagician, 2019-03-04:
not sure but it seems SPIFFS Dir object is not implemented too...but if SPIFFS miss this may be a problem

probonopd, 2019-03-04:
Does this look like what we may need? https://github.com/regishsu/TFT-LCD/blob/6342a179fb09f35c3c8a4932648c08a8af166368/arduino/TFT_eSPI/examples/Smooth%20Fonts/Unicode_test/SPIFFS_functions.ino

GMagician, 2019-03-04:
It seems esp support spiffs like sd....using file to open dir...it may be added to our filewrapper with `#if` but since I don't have an ESP32 i can't do anything on it

GMagician, 2019-03-04:
posted a PR with something...

probonopd, 2019-03-10:
Thanks to your help @GMagician we now have something that compiles on the ESP32 and runs the web interface. I could not make it talk to the printer yet, though.

probonopd, 2019-03-11:
It does talk to the printer now. ESP32 pins used: gpio32 = rx, gpio33 = tx
DO NOT use the normal TX and RX pins one the ESP32 since it messes up the printer's boot process.

closed 2019-03-11 by probonopd

probonopd, 2019-03-11:
SD card not yet tested. Open a new issue if it does not work.