probonopd/WirelessPrinting #30: Cura 3.3.0 "Lost connection with the printer", 2.7.0 worked

probonopd, 2018-04-06:
Cura 3.3.0 "Lost connection with the printer", 2.7.0 worked.

[image]

Maybe we need to implement more of the API for Cura 3.3.0 not to say "Lost connection with the printer"?

cc @fieldOfView

fieldOfView, 2018-04-07:
What do the logs say?

Network printing has had an extensive rewrite in 3.3, and the OctoPrint plugin had to change with it. It is very possible that something broke.

Out of curiosity: why go back to 2.7.0 beta? Since then 2.7, 3.0, 3.1 and 3.2 have been released.

probonopd, 2018-04-07:
I was just comparing the very latest version to the last “known good” combination.... with “logs” you mean the command line output?

probonopd, 2018-04-10:
It seems like what used to be printed to the command line is now hidden away in `$HOME/.local/share/cura/3.3/cura.log`.

```
$ cat ./.local/share/cura/3.3/cura.log
2018-04-10 14:21:23,813 - ERROR - [MainThread] UM.Logger.logException [85]:   File "/home/me/.local/share/cura/3.3/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py", line 302, in requestWrite
2018-04-10 14:21:23,815 - ERROR - [MainThread] UM.Logger.logException [85]:     active_build_plate = Application.getInstance().getBuildPlateModel().activeBuildPlate
2018-04-10 14:21:23,817 - ERROR - [MainThread] UM.Logger.logException [85]: AttributeError: 'BuildPlateModel' object has no attribute 'activeBuildPlate'
```

`'BuildPlateModel' object has no attribute 'activeBuildPlate'` jumps into sight -- is it related to https://github.com/Kriechi/Cura-DuetRRFPlugin/issues/7?

probonopd, 2018-04-10:
Just to confirm, when I click "Print with OctoPrint" nothing happens at all in 3.3.0-BETA whereas the upload was started on 3.2.1.

probonopd, 2018-04-10:
http://192.168.0.16/api/job returns

```
{
  "job": {
    "file": {
      "name": "Some.gcode",
      "origin": "local",
      "size": 113160,
      "date": 1378847754
    },
    "estimatedPrintTime": 8811,
    "filament": {
      "length": 810,
      "volume": 5.36
    }
  },
  "progress": {
    "completion": 0.72,
    "filepos": 820,
    "printTime": 4,
    "printTimeLeft": 547
  }
}
```

http://192.168.0.16/api/printer returns

```
{
  "temperature": {
    "tool0": {
      "actual": 70.66,
      "target": 215.00,
      "offset": 0
    },
    "bed": {
      "actual": 0.0,
      "target": 0.0,
      "offset": 0
    }
  },
  "sd": {
    "ready": 1
  },
  "state": {
    "text": "Operational",
    "flags": {
      "operational": true,
      "paused": false,
      "printing": 1,
      "sdReady": 1,
      "error": false,
      "ready": true,
      "closedOrError": false
    }
  }
}
```

http://192.168.0.16/api/settings is not implemented:

```
$ curl -vvv http://192.168.0.16/api/settings
*   Trying 192.168.0.16...
* TCP_NODELAY set
* Connected to 192.168.0.16 (192.168.0.16) port 80 (#0)
> GET /api/settings HTTP/1.1
> Host: 192.168.0.16
> User-Agent: curl/7.58.0
> Accept: */*
> 
< HTTP/1.1 200 OK
< Content-Length: 2
< Content-Type: application/json
< Connection: close
< Accept-Ranges: none
< 
* Closing connection 0
```

probonopd, 2018-04-10:
When I upload GCODE outside of Cura, then the ESP does not crash and I can now see data in the Monitor of Cura 3.2.1:

[image]

Uploading from Cura 3.2.1 to the same machine (rather than via its web upload form) crashes the ESP.

probonopd, 2018-04-10:
> Uploading from Cura 3.2.1 to the same machine (rather than via its web upload form) crashes the ESP. (Strangely, uploading from Cura 2.7.0 now also crashes it; this was working in earlier builds of WirelessPrinting. Regression? Let's see if 4a7bb16 helps.)

No, it did not help but is a separate issue. Will be opening a new ticket for that.

fieldOfView, 2018-04-10:
I would argue that whatever is supplied to the API, if the ESP crashes it is doing something wrong. Any input supplied through an API should be checked. Returning an error, or ignoring the input if it is not correct is fine, crashing is not.

probonopd, 2018-04-10:
You are right and the crashing is a separate issue. Possibly from coming some HTTP library that I am using.

chepo92, 2018-11-26:
@probonopd which version of cura are you using? i'm in 3.6 now (in windows), monitor says "lost connection with printer" and the controls are unavailable, but print with octoprint button is enabled, and then you can queue the print

probonopd, 2018-11-26:
I have not used this with 3.6 yet. Try 3.0ish-3.4ish. Maybe something has changed that needs to be fixed in the sketch (taking care to keep backward compatibility). 
I find the Cura upload quite finicky, so I end up uploading using the browser most of the time, to be honest. But simply because I don't have enough time to fully debug this and keep it up to all the Cura changes.

probonopd, 2019-02-25:
Cura OctoPrint plugin works with this sketch in Cura 2.7 and 3.0.4, but no longer in 3.5.

Looks like the Cura OctoPrint plugin has changed, and in addition to `printer_state` there is now `print_job_state` which must not be "offline".

This patch is a working workaround for Cura 3.5:

```
sed -i -e 's|print_job_state = "offline"|print_job_state = "ready"|g' $HOME/.local/share/cura/3.5/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py
```

Full diff:

```
me@host:~$ diff -Naur /home/me/.local/share/cura/3.5/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py Desktop/OctoPrintOutputDevice.py 
--- /home/me/.local/share/cura/3.5/plugins/OctoPrintPlugin/OctoPrintPlugin/OctoPrintOutputDevice.py	2019-02-25 14:23:20.121243500 +0100
+++ Desktop/OctoPrintOutputDevice.py	2019-02-25 14:22:15.363841752 +0100
@@ -608,9 +609,8 @@
                             print_job_state = "printing"
                         elif json_data["state"] == "Cancelling":
                             print_job_state = "abort"
-                        elif json_data["state"] == "Operational":
-                            print_job_state = "ready"
-                            printer.updateState("idle")
+                    print_job_state = "ready"
+                    printer.updateState("idle")
                     print_job.updateState(print_job_state)
 
                     print_time = json_data["progress"]["printTime"]
```

@fieldOfView looking at the plugin's source code, this brings me to the question whether `/api/job` needs to report

```
  "state": {
    "text": "Operational",
    "flags": {
      "operational": true,
      "paused": false,
      "printing": false,
      "cancelling": false,
      "pausing": false,
      "sdReady": true,
      "error": false,
      "ready": true,
      "closedOrError": false
    }
```

So far we only have these in `/api/printer`, and according to http://docs.octoprint.org/en/master/api/job.html that is correct.

Is this a bug in the Cura OctoPrint plugin and if so, why is the original OctoPrint not affected by it?

probonopd, 2019-02-25:
Adding `"state":...` to `/api/job` does __not__ make a difference for me.

fieldOfView, 2019-02-26:
The OctoPrint Connection plugin is written to work with OctoPrint. It is regularly updated to work with changes in Cura, and to fix issues that users of the plugin with OctoPrint report. It is great that you have found a different use for the plugin, but this is unintended use and up to you to keep it working.

If you think you have found a bug, and you think your fix does not negatively affect OctoPrint users, feel free to open a pull request. A patch using sed is not accepted. 

From a cursory look at the wall of issue text here and in #52, and from memory about the OctoPrint Connection plugin, I don't think it is a bug in the plugin.

probonopd, 2019-02-28:
The solution is in https://github.com/fieldOfView/Cura-OctoPrintPlugin/pull/105#issuecomment-468212366. Thanks @fieldOfView - I had not considered the possibility of OctoPrint having undocumented API.

closed 2019-02-28 by probonopd