probonopd/WirelessPrinting #31: Gibberish from ESP on serial line can lock up Marlin boot?

probonopd, 2018-04-09:
Gibberish from ESP on serial line can lock up Marlin boot.

When I connect the ESP to a RF100 with Marlin 1.1.8 and power both on simultaneously, the printer's screen flickers for a long time during boot and once Marlin is booted, WirelessPrinting does not work.

If I connect the ESP's TX line half a second or so after having it powered on, then it works.

Is this caused by the gibberish (boot log from the ROM) coming from the ESP during its boot?

Should try https://github.com/esp8266/Arduino/issues/3047#issuecomment-286622666

probonopd, 2018-04-09:
With the serial line attached to D8=GPIO15 as TX and D7=GPIO13 as RX, the ESP does not even boot!

probonopd, 2018-04-09:
With D4 as TX, we are back to the original,  the printer's screen flickers for a long time during boot and once Marlin is booted, WirelessPrinting does not work.

Which makes me wonder whether it's really the serial gibberish that causes this, or the mere fact that the pin is connected...

probonopd, 2018-04-09:
Would a delay() in Marlin "help"?

probonopd, 2018-04-09:
Could this be related/similar to https://github.com/MarlinFirmware/Marlin/issues/10106?

probonopd, 2018-09-08:
For ESP32 there seems to be a way to disable the gibberish:
https://github.com/luc-github/ESP3D/issues/182

chepo92, 2018-11-14:
Had a similar issue of LCD flickering but it was because the 5v regulator of arduino was topped (esp was connected to 5V of ramps), so used a separated regulator for ESP.

In my case gibberish does block boot of Marlin, say, if i turn on ramps and esp at same time they boot correctly, but if I turn on first ramps and 1~2 sec. after turn on the ESP it will block marlin from booting (it doesn't even show splash screen)

probonopd, 2018-11-14:
Since I would like to power the ESP from the printer it would be good to find a way to make this work without having to first switch on the printer and then the ESP.

chepo92, 2018-11-14:
I'm powering esp from ramps board 12v output and a regulator, so they startup at same time, i'll do more testing.

probonopd, 2019-02-28:
I can't reproduce this with the latest build anymore. Closing.

closed 2019-02-28 by probonopd