probonopd/WirelessPrinting #35: got error after enter the ip address of esp8266 ?

fred2088, 2018-04-23:
Dear sir,

got error after enter the ip address of esp8266 (found ip from router) instead of upload gcode page ?

File Not Found

URI: /print
Method: GET
Arguments: 0

Thanks a lot

probonopd, 2018-04-23:
Hi @fred2088 that is strange. How are you trying to print - using one of the slicer applications or entering http://the-ip-address-of-the-printer directly in the browser?

probonopd, 2018-04-23:
`/print` is implemented for POST requests only, but it seems you issued a GET to it:

https://github.com/probonopd/WirelessPrinting/blob/master/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino#L498-L501

fred2088, 2018-04-23:
File Not Found

URI: /
Method: GET
Arguments: 0

any parameter got the same error

fred2088, 2018-04-23:
printer is not yet connected, using windows 7

probonopd, 2018-04-23:
Then it will not work. It needs a printer to be connected because it is querying the printer before starting up.

fred2088, 2018-04-24:
can show printer is not connected if query fail ?

Thanks

probonopd, 2018-04-24:
The way it is currently implemented, the name of the printer is queried from the printer before the rest of the sketch even starts up. If getting the name of the printer does not work, then the system doesn't come up fully. I agree there should be better exception handling but currently I have no time to work on it. Pull requests welcome!

fred2088, 2018-04-25:
which pin i need to connect with mega2560+ramps 1.4 ?

probonopd, 2018-04-25:
RX, TX, GND, +5V, RESET.

fred2088, 2018-04-26:
is the baud rate need to set 115200 and serial0 from marlin ?

probonopd, 2018-04-26:
Yes 115200, for the serial interface you need to set the interface on the printer board to which you physically connect the ESP.

probonopd, 2019-02-28:
Please try with the latest version. It has baudrate autodetection. Does it work now?

closed 2019-03-01 by fred2088