probonopd/WirelessPrinting #36: Problem with Cura

nicstoun, 2018-05-12:
I connect to Cura ,its show  plugin all is perfect  but nothing happened when try print ..  lost connection  I try several times but not success? Check with  with 3.3.0 and 3.3.1 v. not luck

probonopd, 2018-05-12:
Thanks for reporting this issue. Please save the gcode file in Cura, then open a web browser, enter the IP address of your printer (e.g., http://192.168.0.10), and upload the gcode file from your browser.
Does it work then?

For some strange reason, the WirelessPrinting firmware seems to crash when newer versions of Cura upload to it, but it works when the same file is uploaded through the web browser.

I suspect this is is duplicate of https://github.com/probonopd/WirelessPrinting/issues/33, can you confirm?

nicstoun, 2018-05-12:
No,i try and this :  upload is go to 36-40% and then stop with  .page not unavailable error . ..The router is 3 meters from ESP..not sure where is problem. With Octoprint on Raspberry pi print is not problem.

nicstoun, 2018-05-12:
may the problem in module not sure...  :
https://www.ebay.ca/itm/NodeMCU-Lua-ESP-12-WeMos-D1-Mini-WIFI-4M-Bytes-Development-Board-Module-ESP8266-/321989574625
#

probonopd, 2018-05-12:
How large is your gcode file?

probonopd, 2018-05-12:
I don't think that the module is the problem because I am using the same module. Are you using a WEMOS microSD card shield and a microSD card in it?

probonopd, 2018-05-12:
Yes. I think you should try again with a SD shield.

nicstoun, 2018-05-12:
i order and try .,so with Cure problem exist ? 
Thank you !

probonopd, 2019-02-28:
Does it work for you now? I am printing from Cura 3.5 using #5d4ba19 as I write this.

closed 2019-02-28 by probonopd