probonopd/WirelessPrinting #38: Trouble finding library

Baileyjw13, 2018-10-12:
I don't really understand on what I need to do with the 4 HTTPS links. I downloaded them and put them i my Arduino 1.6.9 library folder, but when i open Arduino and look in the    >file                                                                                                                     
                                                                                                             >examples
When i try to upload the sckech to my board i get this error: 

`C:\Users\NUNYOURBUSSINESS\AppData\Local\Temp\Temp1_WirelessPrinting-master.zip\WirelessPrinting-master\ESP8266WirelessPrintAsync\ESP8266WirelessPrintAsync.ino:30:26: fatal error: SPIFFSEditor.h: No such file or directory

 #include <SPIFFSEditor.h>

                          ^

compilation terminated.

exit status 1
Error compiling for board LOLIN(WEMOS) D1 mini Pro.`

Yeah.

Bailey

Baileyjw13, 2018-10-12:
label:closed

probonopd, 2018-10-13:
Looks like the `ESPAsyncWebServer` library is not correctly installed in your Arduino IDE?

closed 2019-02-28 by probonopd