probonopd/WirelessPrinting #43: Upload via Web behavior

chepo92, 2018-11-13:
Uploading gcode via web sends me to http://the-ip-address/api/files/local showing a json file (it doesn't start printing) is that normal behavior? then I have to go to http://the-ip-address/print to start printing

probonopd, 2018-11-13:
I cannot reproduce this. For me the Json is shown and it starts printing. How are you uploading exactly?

chepo92, 2018-11-14:
I'm testing in Windows. Printer is running marlin 1.1.8 
Generate calibration.gcode with cura 3.5.1, (it's correctly generated, tested printing by SD )
(Navigator Firefox 63) Go to respective http://the-ip-address, select file, click upload, printer beeps and says Complete on lcd, browser goes to http://the-ip-address/api/files/local showing json file.
Nothing else happens

chepo92, 2018-11-14:
At some point my sd card is getting corrupted, i'm using a NodeMcu and micro SD shield.