probonopd/WirelessPrinting #45: Saving to Printer SD

just-jason, 2018-12-17:
Marlin 2.0 now has the function of enabling a second serial connection to the board

https://github.com/MarlinFirmware/Marlin/blob/902d5b0f8ba2f46f30fa063b7d2ab8b6193339cc/Marlin/Configuration.h#L110,L116

What are the chances of using this to upload files directly to the Printer SD card/print directly from the SD card?
My interest in this is to make use of the POWER_LOSS_RECOVER FEATURE from Marlin. This however only works on files printed from the Printer SD.

probonopd, 2018-12-17:
The secondary serial line has no other features than the first one. So someone needs to implement

- Open a file on sdcard for writing with Gcode "M28 ".
- Then send the Gcode file contents.
- Close file with "M29".
- Start print with "M23" and "M24" 

However I read [somewhere](https://reprap.org/forum/read.php?219,755123) that depending on the speed of the serial port it can take 10 mins for 3.5 MB. I don't know how far we can increase the speed of the serial port in order to improve performance.

just-jason, 2019-01-05:
Can you tell me, which SD shield are you using?

probonopd, 2019-01-05:
I am using a WeMos D1 mini and a WeMos micro SD card shield.

closed 2019-01-22 by just-jason