probonopd/WirelessPrinting #47: [Question] Doubt about code

GMagician, 2019-01-01:
```c++
  while (serverClient.available()) { // get data from Client
    Serial.write(serverClient.read());
```

shouldn't this while code be:

```c++
  while (serverClient && serverClient.available()) { // get data from Client
    Serial.write(serverClient.read());
```

probonopd, 2019-01-01:
Not sure whether `serverClient` can be false when `serverClient.available())` is true - but why not try and see what happens.

GMagician, 2019-01-01:
I think that until you don't assign it with `serverClient = telnetServer.available();` it may be null so you call an `available` method from a non existent class instance

probonopd, 2019-01-01:
Please try it out and let me know if it works for you, specifically if it makes any difference/improves anything. Thanks for your contribution.

GMagician, 2019-01-01:
Actually I don't have any system ready yet. I'm analyzing code at the moment and I found some things I usually "write" in different ways and I'll propose a PR to see if such "way of" coding may be of your interest

GMagician, 2019-01-01:
I'm also not used to program arduino, it seems like c/c++ but I don't know if exacly the same..for example
`parseTemp` function should return nulll, not "" when temp is not found but maybe arduino compiler is different from my c knowledges

GMagician, 2019-01-01:
And again why don't you use 

```c++
while (!okFound) {
  yield();
```
instead of
```c++
while (okFound == false) {
  yield();
```

other form is compact, but maybe I'm completely in wrong way....

probonopd, 2019-01-01:
Some people think `== false` is more explicit and hence easier to understand, but I am indifferent here

closed 2019-01-01 by GMagician