probonopd/WirelessPrinting #51: [Question] spiffs and write cycles

GMagician, 2019-01-04:
Reading documentation it's said that use of spiffs is for data "almost" invariable I think this is due to limited flash write cycles (I read somewhere about 10000). Is it not too dangerous to use such memory for gcode files? It's true that a file a day is almost 27 years but...

probonopd, 2019-01-04:
I am using an SD card. But yes, if you print a lot then you might need to replace the ESP in 10 years or so. Possibly the same is true, to a lesser extent, to an SD card.

GMagician, 2019-01-04:
SD is about 1000000 and I hope it uses some logic as SDD defices, buty that's ok.
As soon as I tested my code rewamping I'll post a PR. If it's suitable for you you may merge it.

closed 2019-01-04 by GMagician