probonopd/WirelessPrinting #55: Wifi flashing marlin with this method

forum999, 2019-02-17:
sorry not really an issue just a big hope!

Would it be possible to update or reflash my printer with marlin using this method?
this would make things so much easier if i didnt have to keep my old laptop near the printer to make changes and updates to marlin on the 3d printer....

chepo92, 2019-02-19:
You can reflash with an ESP (see esp-link), but not with this repo

forum999, 2019-02-19:
Thanks can you post the link?

chepo92, 2019-02-19:
don't have the link, sending zelda, please follow http://lmgtfy.com/?q=esp-link

just-jason, 2019-02-20:
> sorry not really an issue just a big hope!
> Would it be possible to update or reflash my printer with marlin using this method?
> this would make things so much easier if i didnt have to keep my old laptop near the printer to make changes and updates to marlin on the 3d printer....

Marlin have implemented the function for updating the firmware automatically from the SD card. This is a great looking feature that would open the way to achieve exactly what you are looking for.
https://github.com/MarlinFirmware/Marlin/pull/12043
There is however an issue that makes things a bit more complicated.
The new firmware needs to be on the PRINTER SD CARD, and not on the ESP SD CARD. 
This repository from probonopd has a big advantage in the upload speeds of files to the ESP, but does not save the files to the PRINTER SD CARD. 
There are a few alternatives out there that are able to save files to the PRINTER SD CARD , but I will warn you, the upload speed is VERY SLOW !!!!, and I am not talking slow like I can wait, I am talking slow like banging your head on the wall slow.
This is because of a whole bunch of reasons that many people are trying to find a solution for, but at the moment you have the two options - reasonable upload to ESP SD CARD ( no marlin firmware update possible ) or EXTRA SLOW upload to PRINTER SD CARD ( allows the idea of marlin firmware update )
Give this one a look if you would like to try option two
https://github.com/luc-github/ESP3D

forum999, 2019-02-20:
This looks like something i could work with? Do you have instructionns how to set it up as i cant find anthing in config.h or adv.h reference this in 2.0 bugfix

just-jason, 2019-02-20:
There are some things to set up first 
https://github.com/MarlinFirmware/Marlin/blob/a43e892fb5f180e9ff99815b679b58ff9422deb6/Marlin/Configuration_adv.h#L830
After that you just compile the firmware, save as .bin and transfer to SD card. By the next new start the printer will automatically read and update to the new firmware.

chepo92, 2019-02-20:
With the esp-link bridge you can update directly with no special fw or config,
ESP3D has a bridge implemented 
https://github.com/luc-github/ESP3D

just-jason, 2019-02-20:
> With the esp-link bridge you can update directly with no special fw or config,
OK, I will take your word for it, I thought that the bootloader needed a few mods, but I am glad to hear that it works without.
> ESP3D has a bridge implemented https://github.com/luc-github/ESP3D , if you can work on it to be implemented here would be appreciated
I think implemeting a bridge is not the solution that you are looking for with the current state of development. There is an fundamental problem when using the serial bridge to upload files to the printer SD card in a Marlin based printer It is SLOOOOOOOWWWWWW , really slow !!!. Luc has had many people ask about this issue, and wrote a very good explanation where the problem lies. 
https://github.com/luc-github/ESP3D/issues/292
There is a work around to this, using a WiFi SD card, and can offer a great solution, so the choice is  yours
I think for the moment the solution that Probonopd has here is the most useable with just the ESP and a normal SD card, you will just have to get over the fact that the files are going to end up on the ESP SD card and NOT on the printer SD card. For some cases this is a deal breaker( firmware update - power loss recovery) , I can live with that 8/10.

forum999, 2019-02-21:
Hi so i had a look and added the following to configuration_adv.h in marlin 2.0 bugfix but it refuses to compile:

  /**
   * When using a bootloader that supports SD-Firmware-Flashing,
   * add a menu item to activate SD-FW-Update on the next reboot.
   *
   * Requires ATMEGA2560 (Arduino Mega)
   *
   * Tested with this bootloader:
   *   https://github.com/FleetProbe/MicroBridge-Arduino-ATMega2560
   */
  //#define SD_FIRMWARE_UPDATE
  #if ENABLED(SD_FIRMWARE_UPDATE)
    #define SD_FIRMWARE_UPDATE_EEPROM_ADDR    0x1FF
    #define SD_FIRMWARE_UPDATE_ACTIVE_VALUE   0xF0
    #define SD_FIRMWARE_UPDATE_INACTIVE_VALUE 0xFF
  #endif

  // Add an optimized binary file transfer mode, initiated with 'M28 B1'
//#define FAST_FILE_TRANSFER

Doesn anyone have any ideas

when the compiling fails i get the following errors:

sketch\src\module\planner.cpp: In function '_set_position_mm':

sketch\src\module\planner.cpp:2564:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\secur\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/secur/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

chepo92, 2019-02-22:
> I think implemeting a bridge is not the solution that you are looking for with the current state of development. There is an fundamental problem when using the serial bridge to upload files to the printer SD card in a Marlin based printer It is SLOOOOOOOWWWWWW , really slow !!!. Luc has had many people ask about this issue, and wrote a very good explanation where the problem lies.
> [luc-github/ESP3D#292](https://github.com/luc-github/ESP3D/issues/292)
Yes, but th original question is if it is possible to flash firmware, it is possible with the link, but not implemented, the firm is as much as 256kB so the speed isnt't that terrible, even octoprint does it (and this is kind of a basic port of octoprint for esp). But of course if you want to send a complete gcode file to print you'd better use this repo and write to esp SD, that cost nothing compared to Wifi SD cards, also with wifi sd card you can't command the printer to print the file.

just-jason, 2019-02-22:
> Yes, but th original question is if it is possible to flash firmware, it is possible with the link, but not implemented 

Ok, I will take your word for it, I thought you needed to modify the bootloader, but I am glad to hear that it works without

 also with wifi sd card you can't command the printer to print the file.

This method is being used in other repositories, works great.

forum999, 2019-02-23:
Are there any instructions anywhere for flashing with these methods?

chepo92, 2019-02-23:
> This method is being used in other repositories, works great.

I would like to know those! 

> 
> 
> Are there any instructions anywhere for flashing with these methods?

https://github.com/jeelabs/esp-link

forum999, 2019-02-23:
@justjason

I cant see how to save the marlin build as a bin file? When i goto export binary it saves as .hex files?

just-jason, 2019-02-23:
try this
[image]

forum999, 2019-02-24:
Thanks Jason but it saves as .hex files and not .bin

probonopd, 2019-02-24:
I think hex is fine too?

just-jason, 2019-02-24:
> I think hex is fine too?
 Sorry, I can not tell you much about that. I kind of expect that a Marlin controller can be flashed with the hex, what I am not sure about is weather the process of auto-updating Marlin over SD card is checking for a .bin file to use or not. @forum999 give it a try and let us know how it goes.

forum999, 2019-02-24:
Hex doesnt work? Ive flashed the marlin changes required to config.h

Copied the 2 generated hex  “export compiled binary” files to an sd card root

Put sd into printer

Turned on printer

Nothing happens

probonopd, 2019-02-24:
You are talking about http://domoticx.com/esp8266-wifi-esp-link-firmware/, right? (It is somewhat off-topic here). This might work:

- Use avrdude and point it at port 23 of esp-link. Esp-link automatically detects the programming sequence and issues a reset to the AVR.
- Use avrdude and point it at port 2323 of esp-link. This is the same as port 23 except that the autodectection is not used and the reset happens because port 2323 is used

forum999, 2019-02-24:
Do i need to install a specific boot loader for this? 

As requested the changes i made were:

  //#define SD_FIRMWARE_UPDATE
  #if ENABLED(SD_FIRMWARE_UPDATE)
    #define SD_FIRMWARE_UPDATE_EEPROM_ADDR    0x1FF
    #define SD_FIRMWARE_UPDATE_ACTIVE_VALUE   0xF0
    #define SD_FIRMWARE_UPDATE_INACTIVE_VALUE 0xFF
  #endif

Into configuration.h

Compiled and flashed to printer

just-jason, 2019-02-24:
> Do i need to install a specific boot loader for this?
> As requested the changes i made were:
> //#define SD_FIRMWARE_UPDATE
> #if ENABLED(SD_FIRMWARE_UPDATE)
> #define SD_FIRMWARE_UPDATE_EEPROM_ADDR [...]

I did need to on mine, for SD update. I also have only tried to use it with .bin

probonopd, 2019-07-09:
Something like this might work if we add a transparent serial bridge on port 2323:

` avrdude -DV -p atmega1284p -c arduino -P net:192.168.0.22:2323 -b115200 -U flash:r:factory.hex:i`

But we also need to silence all other communication with the printer if a client connects to that port.
And it will only work if the hardware is connected to the same serial port that is used for flashing the firmware.

probonopd, 2019-07-31:
Updating via the transparent serial bridge did not work for me, and we should probably implement it in a way that is compatible to Octoprint, so that e.g., Cura can be used for updating the firmware...