probonopd/WirelessPrinting #62: Cura does not show temperatures

probonopd, 2019-02-28:
Yay, uploading and printing from Cura 3.5 works for me.

However, Cura does not show the temperatures anymore.

During heating, getting on telnet:

```
me@host:~$ telnet 192.168.0.27
Trying 192.168.0.27...
Connected to 192.168.0.27.
Escape character is '^]'.
< #TIMEOUT#
> G1 X113.952 Y90.635 E0.11222
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X114.542 Y90.533 E0.13462
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X115.009 Y90.509 E0.15212
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X134.991 Y90.51 E0.8998
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X135.59 Y90.55 E0.92226
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X136.177 Y90.669 E0.94467
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X136.744 Y90.865 E0.96712
< New line but nothing to do with it
```

During printing, getting 

```

< ok

  407950
  free heap RAM: 29480

< ok

  407952
  free heap RAM: 26280

< ok

  407954
  free heap RAM: 23080

< ok

  407962
  free heap RAM: 32680

< ok

  407964
  free heap RAM: 29480

> G1 X111.473 Y92.226 E4.04792
< ok

  407969
  free heap RAM: 23080

< ok

  407976
  free heap RAM: 32680

> G0 F3600 X112.067 Y92.448
< ok

  407989
  free heap RAM: 35880

> G1 F1800 X112.463 Y92.055 E4.0688
< ok

  408001
  free heap RAM: 32680

> G1 X112.911 Y91.721 E4.08971
< ok

  408012
  free heap RAM: 35880

> G1 X113.402 Y91.456 E4.11058
< ok

  408055
  free heap RAM: 35880

> G1 X113.926 Y91.263 E4.13148
```

probonopd, 2019-02-28:
It seems like `M105` is not being sent to the printer anymore. Hecne we are not getting temperature from the printer anymore.

probonopd, 2019-02-28:
Probably caused by 

>     * added support to Cap:AUTOREPORT_TEMP (don't send M105 but only M155 on connection established)

in https://github.com/probonopd/WirelessPrinting/pull/52

cc @GMagician

probonopd, 2019-02-28:
What was wring with sending `M105`? It seemed to have worked reliably.

probonopd, 2019-02-28:
We are getting

```
> M105
< ok T:198.9 /200.0 B:60.0 /60.0 T0:198.9 /200.0 @:81 B@:72
```

but it is not parsed correctly?

chepo92, 2019-02-28:
Because my PR fixes it https://github.com/GMagician/WirelessPrinting/pull/9

chepo92, 2019-02-28:
> 
> 
> We are getting
> 
> ```
> > M105
> < ok T:198.9 /200.0 B:60.0 /60.0 T0:198.9 /200.0 @:81 B@:72
> ```
> 
> but it is not parsed correctly?

I think your printer doesn't have autotemp respones cap enabled, i figured and solved it yesterday. I'll make PR directly to master

GMagician, 2019-02-28:
It will never be parsed because of ok in front of M105...I'm going to fix it

chepo92, 2019-02-28:
#67 Fixes it

probonopd, 2019-02-28:
Thanks @GMagician, 1740455 fixes it for me :+1:

PR #64 (2019-02-28): Some missing fix
Fix #62 and some fixes chepo92 sent and not merged in time
closed 2019-02-28 by probonopd