probonopd/WirelessPrinting #63: Implement /api/printer/command

probonopd, 2019-02-28:
We now can implement `/api/printer/command` for using the Cura controls.

[image]

chepo92, 2019-02-28:
When I saw the Lost connection with printer gone I tried tho buttons I realised we haven't implemented it yet. I'll do it, pls assign it to me.

GMagician, 2019-03-02:
@chepo92 I'm working at json6 and this will come with it.... something is already present but not fully working

probonopd, 2019-03-02:
Regading json6, are you sure that _all_ the libraries we are using can handle it?

GMagician, 2019-03-02:
I'm already able to send gcode and pause printer but more test must be done

closed 2019-03-03 by probonopd

probonopd, 2019-03-03:
Thanks @GMagician

commit 3492ecdd (2019-03-03): Update ArduinoJSON and implement /api/printer/command

Closes #63, closes #69