probonopd/WirelessPrinting #68: Implement Web Server OTA (https)

probonopd, 2019-02-28:
Currently it is too complicated to update the firmware. It would be nice if one could update from the HTTP interface directly with the firmware being downloaded from a https link.

- https://github.com/esp8266/Arduino/issues/1021
- https://gist.github.com/igrr/24dd2138e9c8a7daa1b4
- ESPhttpUpdate.update(String(host),httpsPort,String(url),String(""),String(fingerprint));

probonopd, 2019-03-02:
Could also implement uploading the firmware via a web form:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/35

probonopd, 2019-03-02:
> We should get as the default firmware URL the latest GitHub Release, from https://api.github.com/repos/probonopd/WirelessPrinting/releases [0]["assets"][0]["browser_download_url"]

@yknivag seems to have been working on this at
https://github.com/yknivag/test/blob/master/temp/temp.ino

Also see
https://gist.github.com/yknivag/2707e27787abeafa69b33c4e9d043c3a

ayushsharma82, 2019-04-24:
Any thoughts about using ElegantOTA ? It will provide a pleasant UI for uploading OTAs. 

https://github.com/ayushsharma82/ElegantOTA

probonopd, 2019-04-24:
Thanks for the pointer @ayushsharma82. I was not aware of ElegantOTA so far but it does look elegant indeed.

* We are using AsyncHTTPServer in this project. Would this work for ElegantOTA?
* Would using ElegantOTA provide any benefit over our current web OTA plus using ESP-DASH for a better UI?

ayushsharma82, 2019-04-24:
It currently works with default Webserver of ESP. But it should take about an hour to port it to AsyncWebServer. I will develop a Async Version Soon. 

Currently there is just one benefit, It shows progress of OTA via a progress bar on webpage and shows a good and clear status when OTA is done. ElegantOTA is certainly for beauty purposes only. 😄

probonopd, 2019-04-24:
> I will develop a Async Version Soon.

Cool. Please let me know when it is available.

> It shows progress of OTA via a progress bar

Definitely an advantage.

Does updating work using the command line (e.g., wget/curl) too, like our current solution?

ayushsharma82, 2019-04-24:
Yes, In fact the webpage does the same but with a AJAX Request to a upload URL.

ayushsharma82, 2019-05-02:
@probonopd  Just Released AsyncElegantOTA , It works on AsyncWebServer and has support for both esp8266 and esp32. 

https://github.com/ayushsharma82/AsyncElegantOTA

probonopd, 2019-05-02:
@ayushsharma82 that's great news. Would you like to give it a try to integrate it into WirelessPrinting?

ayushsharma82, 2019-05-02:
@probonopd  Created a Pull Request. Edit travis cli to copy AsyncElegantOTA repo for successful builds.  https://github.com/probonopd/WirelessPrinting/pull/118

yknivag, 2019-11-13:
Apologies for the very late reply, but I think I now have this working.  See https://github.com/yknivag/ESP_OTA_GitHub

probonopd, 2019-11-14:
Wow this is useful. Would you like to send a pull request to this repository incorporating it @yknivag?

yknivag, 2019-11-14:
Thank you for your appreciation and excitement. :-)

I only just got it working yesterday after waiting 6 months for a bug fix to 
the ESPhttpUpdate library.

I think maybe a some more thorough testing is in order before I would be presumptuous enough to submit a PR with it to someone else's project.  It's definitely a beta atm!

Please do feel free to test it out though, but I would keep it in your dev branch until there is a release on the repo.

probonopd, 2022-04-26:
@yknivag is it still working and being maintained?

yknivag, 2022-04-27:
@probonopd It was still working last time I checked it.  I haven't had cause to use it myself for a little while due to working on other things, but I see no reason why the library shouldn't still work.