probonopd/WirelessPrinting #71: Provide a way to print when "Print Immediately" was not selected

probonopd, 2019-02-28:
Provide a way to print when "Print Immediately" was not selected

chepo92, 2019-03-02:
maybe in the same root page display the last uploaded file and a print button?

probonopd, 2019-03-02:
Yes, that would be good.

probonopd, 2019-03-03:
Instead of adding another endpoint, we could use the one we already have, `/api/job`.

According to http://docs.octoprint.org/en/master/api/job.html, one must send

```
POST /api/job HTTP/1.1
Host: example.com
Content-Type: application/json
X-Api-Key: abcdef...

{
  "command": "start"
}
```

Now, how can we POST the above from a form/by clicking a button in HTML?

Put it in a hidden textarea and pass to server?

probonopd, 2019-03-03:
Looking at https://stackoverflow.com/a/6418506, 

```
var xmlhttp = new XMLHttpRequest();   // new HttpRequest instance 
xmlhttp.open("POST", "/api/job");
xmlhttp.setRequestHeader("Content-Type", "application/json");
xmlhttp.send(JSON.stringify({command:"start"}));
```

commit e7affe31 (2019-03-03): Implement buttons, closes #71
closed 2019-03-03 by probonopd