probonopd/WirelessPrinting #72: Print stalled. Printer displays "M105" on the LCD screen

probonopd, 2019-02-28:
When uploading the attached Slic3r-generated GCODE using the web interface (not using Cura):

```
me@host:~$ telnet 192.168.0.27
Trying 192.168.0.27...
Connected to 192.168.0.27.
Escape character is '^]'.
< #TIMEOUT#
> G80 
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 Y-3.0 F1000.0 
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G92 E0.0
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X60.0 E9.0  F1000.0 
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G1 X100.0 E12.5  F1000.0 
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> G92 E0.0
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> M900 K30
< New line but nothing to do with it
< T:211.88 E:0 B:59.8
ok

  57420
  free heap RAM: 36216

> G21 
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> M105
< New line but nothing to do with it
< T:216.7 E:0 W:0
ok

  61076
  free heap RAM: 36216

> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< ok

  68347
  free heap RAM: 36232

> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< New line but nothing to do with it
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< New line but nothing to do with it
< Go home finished
ok

  94554
  free heap RAM: 33016

< New line but nothing to do with it
< Full RX Buffer
ok

  94557
  free heap RAM: 26616

< ok

  94559
  free heap RAM: 23432

< ok

  94570
  free heap RAM: 33032

< ok

  94572
  free heap RAM: 29832

< ok

  94574
  free heap RAM: 26632

< New line but nothing to do with it
< Invalid M code.
ok

  94584
  free heap RAM: 33016

< ok

  94586
  free heap RAM: 29832

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94589
  free heap RAM: 26584

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94591
  free heap RAM: 23384

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94599
  free heap RAM: 32984

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94601
  free heap RAM: 29784

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94604
  free heap RAM: 26584

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94607
  free heap RAM: 23384

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94616
  free heap RAM: 32984

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94619
  free heap RAM: 29784

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94621
  free heap RAM: 26584

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94624
  free heap RAM: 23384

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94632
  free heap RAM: 32984

< ok T:212.5 /215.0 B:60.5 /60.0 T0:212.5 /215.0 @:127 B@:0

  94634
  free heap RAM: 29784

> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
(repeats forever)
```
The printer stalls right at the end of the start (header) GCODE.
The printer displays "M105" on the LCD screen.

[GCODE.zip](https://github.com/probonopd/WirelessPrinting/files/2917341/GCODE.zip)

probonopd, 2019-02-28:
Interestingly, the same seems to print just fine when using Slic3r PE to upload... what gives?

chepo92, 2019-03-01:
What firmware do you have?  I don't see G80 documented in marlin

probonopd, 2019-03-01:
Prusa Original i3 MK2S 3.0.12-RC2
https://shop.prusa3d.com/forum/others-archive--f82/firmware-3-0-12-rc2-mk2-multi-material-mk2-s-mk1-t4946.html

probonopd, 2019-03-01:
The exact same GCODE prints without a hitch on 4b725ce when using `curl -F "file=@/path/to/some.gcode" -F "print=true" 192.168.0.27/api/files/local`. More investigation needed.

chepo92, 2019-03-01:
I think "New line but nothing to do with it" should be followed by the message received for better debug.
Also it could be useful to implement firmware detection.

https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes
I don't see M900

><Connected to 192.168.0.27.
><Escape character is '^]'.
> <#TIMEOUT#

It'starts timing out

chepo92, 2019-03-01:
> 
> 
> The exact same GCODE prints without a hitch on [4b725ce](https://github.com/probonopd/WirelessPrinting/commit/4b725cef20c3633ba575d94203cae587377b9558) when using `curl -F "file=@/path/to/some.gcode" -F "print=true" 192.168.0.27/api/files/local`. More investigation by myself needed.

Maybe something with the upload form.

probonopd, 2019-03-01:
> I don't see M900

That is Linear Advanced, a very cool feature:
https://help.prusa3d.com/article/t5w9VsdVai-linear-advanced

GMagician, 2019-03-01:
> I think "New line but nothing to do with it" should be followed by the message received for better debug.
Also it could be useful to implement firmware detection.

firmware detection may be done in response to M115.

chepo92, 2019-03-01:
> 
> 
> > I don't see M900
> 
> That is Linear Advanced, a very cool feature:
> https://help.prusa3d.com/article/t5w9VsdVai-linear-advanced

Now I found it http://marlinfw.org/docs/gcode/M900.html

#77 Will help us here

chepo92, 2019-03-01:
> firmware detection may be done in response to M115.

Yes, but how many do we address? Octoprint does fw detection, maybe start from there. 
But what then? this issue has to do with special responses of the firmware. eg:

> < New line but nothing to do with it
> < Go home finished
> ok

probonopd, 2019-03-01:
Confirmed, same GCODE stalls after the start GCODE with

```
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
```

when uploaded through the web form.

probonopd, 2019-03-01:
The uploaded file on the SD card is identical to the original:

```
me@host:~$ md5sum some.gcode 
3447c562d334301f2cac99d6bc4a7fd8  some.gcode # original file on the computer
me@host:~$ md5sum '/media/me/691B-FBBE/some.gcode'
3447c562d334301f2cac99d6bc4a7fd8  /media/me/691B-FBBE/some.gcode # uploaded to the SD card using the web form
```

chepo92, 2019-03-01:
> 
> 
> Confirmed, same GCODE stalls after the start GCODE with
> 
> ```
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> > M105
> < #TIMEOUT#
> ```
> 
> when uploaded through the web form.

You have print inmediately checed right? 
I experience some esp crash and reboot sometimes after cura upload, did it not crash ?

chepo92, 2019-03-01:
@probonopd Can you telnet it before uploading? (and before it starts timingout)

probonopd, 2019-03-01:
Same result with SPIFFS (without SD card).

probonopd, 2019-03-01:
Here is a fulll log from before I upload to the point where it stalls.

The download is complete (printer beeps two times) around `M190 S60`.

__To my suprise this time it works without stalling.__

I suspect it has something to do with the `New line but nothing to do with it` and `#TIMEOUT#`.

GMagician, 2019-03-01:
I think that the real big problem is that when timeout is detected it purge command from queu but if command is really not yet terminated, a new command will be sent (and lost by printer).
Maybe: 
1) prevent timeout to be raised refining restonses handling
2) on timeout stop printing, purge whole queue and set error

probonopd, 2019-03-01:
Are we 100% sure that we are ever only sending anything after we have received a line that starts with "ok"?

This is what I had in the sketch when it still worked for me:

```
String sendToPrinter(String line) {

  /* Although this function does not return before okFound is true,
     somewhere else in the sketch this function might also have been called
     hence we make sure we have okFound before we start sending. */

  while (okFound == false) {
    yield();
  }

  Serial.setTimeout(240000); // How long we wait for "ok" in milliseconds
  (...)
```

chepo92, 2019-03-01:
From your log: 

> > M190 S60 
> < New line but nothing to do with it
> <T:130.52 E:0 B:49.9
> 
> < New line but nothing to do with it
> <T:130.52 E:0 B:49.9
> T:129.91 E:0 B:49.9

Your firm temp response doesn't have a leading space before T (" T:"), so it is not parsing the temp then not resting timeout. If we delete that space the M115 will probably break, so we need to make temp parser more robust but also M115

chepo92, 2019-03-01:
Maybe a variable "printerConnected" would do the trick, so temp parser (or any other parser) doesn't mess with M115 reply

GMagician, 2019-03-01:
better a more robust parser...

GMagician, 2019-03-01:
is it possible to have a new "telnet" log with new PRs? Starting from very first printed line

probonopd, 2019-03-01:
In the middle of a long print right now... next time, sure

probonopd, 2019-03-01:
During printing I don't get any unhandled lines or timeouts anymore with 1094f38. Only before the print starts.

GMagician, 2019-03-01:
yes but how long have we to wait? We can try to use temp and position but this will not tell us command has been received by printer, only that printer is not dead.
I posted a PR with relaxed timeout check (original chepo92 solution) even if I don't like it so much I think there is no better solution.
With my marlin version I have a lot of busy messages when homing, heating and pausing but it seems Prusa MKx is a little bit "lazy" on this

probonopd, 2019-03-01:
Let's not assume too much what the printer does nor does not do - there are many different firmware flavors... "ok" means that the printer is ready for the next command iirc.

GMagician, 2019-03-01:
yes if you see your temp feedback when you get "unhandled line" you will see that this is not a stantard temp response...it misses target temps

probonopd, 2019-03-01:
Attached is a log from 83dea71. We are still sending M105 when there is no `ok`. Consequently we still get `<Full RX Buffer` during heating.

The 

```
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
< #TIMEOUT#
> M105
```

is while the printer performs mesh level calibration.

It __still stalls__ after the intro GCODE and shows "M105" on the LCD.

[log.txt.zip](https://github.com/probonopd/WirelessPrinting/files/2921425/log.txt.zip)

probonopd, 2019-03-01:
Looking at the log, it seems quite obvious to me what is going on:
We are sending `M190 S60` to heat the bed to 60 degrees. This takes minutes before the printer responds with ok. During this time we are not allowed to send further commands to the printer but we are doing so since it says `#TIMEOUT#` and then continues to send commands.

commit c970d4a7 (2019-03-02): Cleanup serial handling and logs (#87)

Closes  #72, closes #84 (with the addition of #tags#), closes #85
closed 2019-03-02 by probonopd