probonopd/WirelessPrinting #74: G28 Timing Out

chepo92, 2019-03-01:
I'm getting this: 
```
> M82 
< ok

  391624
  free heap RAM: 32096

> M107 
< ok

  391635
  free heap RAM: 30736

> G28 X0 Y0 
< AutoReportTemps parsed
< #TIMEOUT#
> G28 Z0 
< AutoReportTemps parsed
< MPosition parsed
< ok

  396167
  free heap RAM: 34096

> 
< AutoReportTemps parsed
< #TIMEOUT#
> G1 Z15.0 F9000 
< AutoReportTemps parsed
< MPosition parsed
< ok

  400953
  free heap RAM: 34080
```

GMagician, 2019-03-01:
```c++
if (lastCommandSent.startsWith("G28"))
  restartSerialTimeout();   // Some firmware doesn't send busy while homing but just position. It can be used to prevent timeout
```
should prevent this

GMagician, 2019-03-01:
and.... when a timeout is got messages is removed .. subsequent ok may desync code

chepo92, 2019-03-01:
> 
> 
> ```c++
> if (lastCommandSent.startsWith("G28"))
>   restartSerialTimeout();   // Some firmware doesn't send busy while homing but just position. It can be used to prevent timeout
> ```
> 
> should prevent this but why doesn't it send position like it always did?

It does
```
> G28 X0 Y0 
< AutoReportTemps parsed
< #TIMEOUT#
> G28 Z0 
< AutoReportTemps parsed
< MPosition parsed
< ok
```

But it send it after the G28 homing is completed, and then an ok, also in betweeen it sends autotemps

```

      else if (parsePosition(serialResponse)) {
        telnetSend("< MPosition parsed");
        if (lastCommandSent.startsWith("G28"))
          restartSerialTimeout();   // Some firmware doesn't send busy while homing but just position. It can be used to prevent timeout
        GotValidResponse();   // Warning, this will empty 'serialResponse'
      }
```

It shouldn't be inside the  if (parsePosition(serialResponse)) { ...  but outside. I see what you did here... (remember long running commands)

GMagician, 2019-03-01:
well I think that maybe is not a good solution to be so accurate to reset timeout only on specific commands...I really think that your original solution will relief from pain

GMagician, 2019-03-02:
@chepo92 this has been fixed (timeout is still present on longer operation but don't go on sending other data)

probonopd, 2019-03-03:
@chepo92 can you confirm that it is working for you? Then we can close this issue.

chepo92, 2019-03-03:
confirm, still timing out but not sending more gcode until ok received.

probonopd, 2019-03-03:
`#TIMEOUT#` just means that the printer did not send anything for the last 2.5 seconds which may be perfectly normal. What is important, though, is that we don't send further GCODE before the printer responds with `ok`. (My printer does not respond for minutes while calibrating.)

chepo92, 2019-03-03:
```
> > M107 
> < ok
> 
>   391635
>   free heap RAM: 30736
> 
> > G28 X0 Y0 
> < AutoReportTemps parsed
> < #TIMEOUT#
> > G28 Z0 
> < AutoReportTemps parsed
> < MPosition parsed
> < ok
> 
>   396167
>   free heap RAM: 34096
> 
> > 
> < AutoReportTemps parsed
> < #TIMEOUT#
> > G1 Z15.0 F9000 
> < AutoReportTemps parsed
> < MPosition parsed
> < ok
```

probonopd, 2019-03-03:
Which version of the sketch are you running? It should be shown at the bottom of the `/` page in recent versions.

chepo92, 2019-03-03:
2.x-localbuild, I'm synced with master
https://github.com/probonopd/WirelessPrinting/compare/master...chepo92:master

GMagician, 2019-03-03:
As stated in another post... heap status is no more reported, nor < #timeout#. This seems not updated bin

GMagician, 2019-03-03:
@chepo92 looking at yours PR it seems you are not perfectly synced with master...there are a lot or "old" pr inside your fix

chepo92, 2019-03-03:
Working

closed 2019-03-03 by chepo92