probonopd/WirelessPrinting #81: [Question] versioning with travis

GMagician, 2019-03-01:
New change to update version with travis will prevent local compile to get a valid program

probonopd, 2019-03-01:
What error are you getting?

GMagician, 2019-03-01:
XXXVERSIONXXX because of no local travis

probonopd, 2019-03-01:
Sure, this is the placeholder where the sed command in .travis.yml would put in the git hash. But does it actually prevent from being compiled?

GMagician, 2019-03-01:
No sorry wrong statement. It compile but cura and html report an invalid version.
Any user can't have a "correct" versioned version this way without local travis

probonopd, 2019-03-01:
Yes, because we don't know the version when it is compiled locally... we could use a different placeholder though. Ideas?

GMagician, 2019-03-01:
that may be better than current

probonopd, 2019-03-01:
Can you check https://transfer.sh/9bOms/ESP8266WirelessPrintAsync_d1_mini_988efa7.bin please?

GMagician, 2019-03-01:
How can I download it? .. with arduino ide 

N.B. you may also remove `#define VERSION`. It's no longer used

probonopd, 2019-03-01:
See the README.md for uploading. You can upload using Python.

commit aa88c9fc (2019-03-01): 2.x-localbuild, fix #81

[ci skip]
PR #83 (2019-03-01): Fix #81, use other version for local builds
Use `2.x-localbuild`  as the placeholder for the git commit when doing local builds, fixes #81
commit 83dea717 (2019-03-01): Fix #81, use other version for local builds (#83)
closed 2019-03-01 by probonopd