probonopd/WirelessPrinting #84: Clean up telnet log

probonopd, 2019-03-01:
* All lines coming from the printer should start with `< `
* All lines sent to the printer should start with `> `

I do not understand what the following means:

```
Unhandled line received:`// This does not start with < or > because it is a comment
<T:41.52 E:0 B:29.1 // This has been received by the printer
T:44.16 E:0 B:29.1 // WHAT IS THIS?
T:47.67 E:0 B:29.5 // WHAT IS THIS?
```

GMagician, 2019-03-02:
ok.. that's a weird thing in prusa firmware...
```c++
		KEEPALIVE_STATE(NOT_BUSY);
        while ( (target_direction)&&(!cancel_heatup) ? (isHeatingBed()) : (isCoolingBed()&&(CooldownNoWait==false)) )
        {
          if(( millis() - codenum) > 1000 ) //Print Temp Reading every 1 second while heating up.
          {
			  if (!farm_mode) {
				  float tt = degHotend(active_extruder);
				  SERIAL_PROTOCOLPGM("T:");
				  SERIAL_PROTOCOL(tt);
				  SERIAL_PROTOCOLPGM(" E:");
				  SERIAL_PROTOCOL((int)active_extruder);
				  SERIAL_PROTOCOLPGM(" B:");
				  SERIAL_PROTOCOL_F(degBed(), 1);
				  SERIAL_PROTOCOLLN("");
			  }
				  codenum = millis();
			  
}
```
if set system not busy and when open a loop waiting for temp...and it report a custom temp message

WOW it is busy to wait temp but it set a no busy state....that may be defined as pure logic!

probonopd, 2019-03-02:
Why are some lines not prefixed with "<"? Are they not coming from the printer?

GMagician, 2019-03-02:
yes since there is no ok and is not parsed as a know line it goes to "unhandled"

GMagician, 2019-03-02:
..and there no timeout reset...
I'm working now to fix this and other issue

probonopd, 2019-03-02:
We should make the code more robust by printing _everything_ coming from the printer to telnet, prefixed with `<`, and reset the "alive" timer. Regardless of what it is. And only _thereafter_ try to do something with it. If we can't do something with the line, we could say "Last line could not be interpreted".

commit c970d4a7 (2019-03-02): Cleanup serial handling and logs (#87)

Closes  #72, closes #84 (with the addition of #tags#), closes #85
closed 2019-03-02 by probonopd