probonopd/WirelessPrinting #85: Delay sending anything to the printer until ok is received

probonopd, 2019-03-01:
This is how I was doing it in [1.x](https://github.com/probonopd/WirelessPrinting/tree/1.x) and I had never any trouble with it.

```
bool okFound = true; // Set to true if last response from 3D printer was "ok", otherwise false

(...)

String sendToPrinter(String line) {

  /* Although this function does not return before okFound is true,
     somewhere else in the sketch this function might also have been called
     hence we make sure we have okFound before we start sending. */

  while (okFound == false) {
    yield();
  }

  Serial.setTimeout(240000); // How long we wait for "ok" in milliseconds

  (...)

  Serial.println(line); // Send to 3D Printer

  (...)

  lineLastSent = line;
  okFound = false;
  while (okFound == false) {
    response = Serial.readStringUntil('\n');
    parseTemperatures(response);
    lineSecondLastReceived = lineLastReceived;
    lineLastReceived = response;
    if (response.startsWith("ok")) okFound = true;
  }
  return (response);
}
```

probonopd, 2019-03-01:
In essence, we are not sending anything to the printer before the previous response started with OK.

GMagician, 2019-03-02:
I usually don't like the toooo long wait for timeout...
I think that we have to detect if printer is live and if communication is still live and have calibrated timeout for it.
Going back to your issues during heating, they are because it seems your printer sent a different message when it is heating, it sends only current temp. Since temp format is different it is not parsed and it will timeout.
I prefer to add anoter timeoutreset in "unhandled" if branch. In that case I reset timer because communication is live and printer is live....
but this is just my idea

probonopd, 2019-03-02:
Can you implement waiting forever on ok in a branch, for testing purposes?

probonopd, 2019-03-02:
I think we need to distinguish two different things:
* Checking whether the printer is still live - whenever _anything_ is received from the printer we can reset the timer that checks whether the printer is still alive (question: what do we do if it isn't?)
* Assuming that we are allowed to send new data to the printer - this is only after the printer has responded with `ok` 

Does this make sense?

GMagician, 2019-03-02:
I'm working on it

commit c970d4a7 (2019-03-02): Cleanup serial handling and logs (#87)

Closes  #72, closes #84 (with the addition of #tags#), closes #85
closed 2019-03-02 by probonopd