probonopd/WirelessPrinting #91: Switch off heated bed when we are at 5mm Z or above

probonopd, 2019-03-02:
Why are most people heating the bed until the print has finished? With most models on most printers, keeping the bed heated after the print has already reached a certain Z position does not add value, only costs energy and adds time for having to wait for the bed to cool down before the print can be removed.

So personally I am always switching off the bed heating after the print has reached, say, 5mm Z height.

Some slicers can insert custom GCODE at certain layer heights but it would be nice if we could implement this feature in WirelessPrinting, so that it would work with all slicers and would not require special slicer configuration.

Pseudo-code:

if last line sent to printer starts with `G1 Zx` with `x >= 5`, then insert a `M140 S0` command to turn off the heatbed.

GMagician, 2019-03-02:
Personally I don't like to interfere with sent gcode.

just-jason, 2019-03-02:
> So personally I am always switching off the bed heating after the print has reached, say, 5mm Z height.

Do you ever have difficulties after you do this with print adhesion ?

probonopd, 2019-03-02:
No. Print adhesion is great when making the first few layers, and once you have reached 5mm height (the exact value can be tuned dependent on other parameters) then the material between the first layer and the layer at 5mm height is cooled down enough such as to be stiff enough so that the object cannot warp anymore. At that point you can switch off the heatbed.

Advantages:
* Less energy consumption
* Can remove prints immediately when they are complete, no need to wait
* No burning fingers when removing prints :+1: 

I have been doing this for months. (I am mostly using PLA, so the exact height needed might be different for other materials but the principle should be the same.)

chepo92, 2019-03-02:
You are right if you are using PLA, but if you are printing ABS (or other warping material) with a closed chamber the heat bed keeps the room temperature stable to avoid warping. 
IMHO the way to do this is in the slicer (Custom gcode, script, etc) not the "host".

just-jason, 2019-03-05:
There is an extension in CURA - ChangeAtZ, that will do this for you in the slicer. It could be a solution 
[image]

probonopd, 2020-03-17:
Closing for now since this can be done in the slicer, feel free if you think this should be reopened.

closed 2020-03-17 by probonopd