probonopd/WirelessPrinting #94: Add camera support

probonopd, 2019-03-02:
This is blocked by #29.

There are now **ESP32-CAM** modules with camera and SD card. Can be had for around USD 10 from China. Ideal for our purpose.

[image]
[image]

@me-no-dev has added Arduino support for it :
https://github.com/espressif/esp32-camera/issues/7#issuecomment-449624965

It seems to be working well:
> Thank you so much, worked out of the box here as well. I tried it on an ESP32-CAM with a wide angle lense I got from a TTGO T-Journal order.

chepo92, 2019-03-03:
Wow, buying it !

probonopd, 2019-03-11:
Internal pin connections:

[image]

Source:
https://github.com/SeeedDocument/Outsourcing/raw/master/113990580%20ESP32-CAM/113990580%20ESP32-CAM%20Product%20Specification.pdf

probonopd, 2019-03-11:
We may not have free pins to use Serial1 on this board? GPIO16 seems to be the only pin that is on the board and not connected to cam or SD.

chepo92, 2019-03-11:
> We need to find a way to not get the ESP32 boot messages onto the serial line since they will make it impossible to boot the printer firmware (I have tested it!).

I'll test it too 

> We may not have free pins to use Serial1 on this board? GPIO16 seems to be the only pin that is on the board and not connected to cam or SD.

I haven't tested but I think Uarts can be mapped to any pin: https://quadmeup.com/arduino-esp32-and-3-hardware-serial-ports/

probonopd, 2019-03-11:
Yes, but you need free pins that are not connected to the camera nor the SD card. And it looks like we are short of those...

probonopd, 2019-03-11:
In addition to making a video stream of the printer, and taking timelapse images, we might do QR code recognition to e.g., recognize settings for different filaments based on QR codes.

https://github.com/donny681/ESP32_CAMERA_QR/

GMagician, 2019-03-11:
I read somewhere that boot log messages may be disabled on esp32 and not on esp8266, but I forget where I read it

fred2088, 2019-03-23:
buy one, will test it out ...

muenan, 2019-03-27:
> 
> 
> Yes, but you need free pins that are not connected to the camera nor the SD card. And it looks like we are short of those...

Maybe you can use the LED PIN (GPIO33) and with SoftSerial (RX, TX | GPIO16, GPIO33). At the LED PIN you can solder a wire to the + side for the TX. The user (v12345vtm : https://github.com/v12345vtm/timelapse-ESP32-CAM ) has developed a timelapse feature. If you could put the project and his together, it would be ideal.

probonopd, 2019-03-27:
Let's try it out. Unfortunately I don't have this hardware.

fred2088, 2019-05-17:
we bought it and would like to try it out ...

probonopd, 2019-05-17:
Looking forward to hearing from you how it goes @fred2088

probonopd, 2019-07-06:
Have you made any progress @fred2088?

fred2088, 2019-07-06:
thinking to merge two project 
https://github.com/v12345vtm/timelapse-ESP32-CAM

ETE-Design, 2019-09-21:
So we continue here... Can't help coding :-( My skills is not there yet... But have you looked at this project? Mabye it can help?
https://randomnerdtutorials.com/esp32-cam-video-streaming-web-server-camera-home-assistant/

probonopd, 2019-09-21:
> Can't help coding :-( My skills is not there yet...

It's just Arduino code - so not that complicated really. Probably mostly a matter of figuring out which pins to use.

> BTW, do you have some screenshots of your GUI?

No, why? Just upload it to your ESP board and check for yourself...

ETE-Design, 2019-09-21:
The pins is here?
https://github.com/RuiSantosdotme/ESP32-CAM-Arduino-IDE/blob/master/ESP32-CAM-Video-Streaming/ESP32-CAM-Video-Streaming.ino

Was just asking for screenshots in general, just so more people than me can see what this is and compare to ESP3D without the need og flashing...

probonopd, 2019-09-21:
> The pins is here?

Yes, maybe. Just try it out until you have it workning.

> Was just asking for screenshots in general, just so more people than me can see what this is and compare to ESP3D without the need og flashing...

Check this video:

https://www.youtube.com/watch?v=4YOg0ReJtec

ETE-Design, 2019-09-21:
So no Web GUI only direct integrated to Cura? Was hoping to have something like ESP 3D, not only direct control from Cura...

probonopd, 2019-09-21:
There is also a Web interface to upload and print Gcode.

ETE-Design, 2019-09-21:
Okey, just not show in the video then :-)

ETE-Design, 2019-09-21:
Just now I only have the ESP8266 to Play around with... Waiting for it to be delivered from Ali...

probonopd, 2019-09-21:
ESP8266 is fine too!

ETE-Design, 2019-09-21:
Just dont have a SDCard connected direct to it...

probonopd, 2019-09-21:
No need to, you can also use SPIFFS

probonopd, 2019-10-04:
> We need to find a way to not get the ESP32 boot messages onto the serial line since they will make it impossible to boot the printer firmware (I have tested it!).

Actually, this is easy. Tested with ESP32-CAM firmware.

* Connect pin UOR to PL2303HX green = TX
* Connect pin UOT to PL2303HX white = RX
* Connect pin 5V to PL2303HX red = 5V
* Connect  pin GND to PL2303HX black = GND
* Connect pin IO15 to GND. __Boot messages gone__

ETE-Design, 2019-10-04:
So now that you got it booting there is not that Long way to make the cam Work also :-)

probonopd, 2019-10-04:
Someone just needs to do it.

Colin66, 2019-10-21:
Hi, there's a project interfacing esp32-cam to 3d printer which appears relevant here :

https://github.com/eben80/ESP32-CAM-Print-Monitor

probonopd, 2019-10-22:
Hi @Colin66 very good, so it looks like @eben80 has figured out how to access the camera while also talking to the printer via the serial line. Hence it should be a breeze to combine it with WirelessPrinting. @Colin66 would you be interested?

Colin66, 2019-10-22:
Hi @probonopd I'm interested in the output but unfortunately I don't have the coding skills.

eben80, 2019-10-23:
> Hi @Colin66 very good, so it looks like @eben80 has figured out how to access the camera while also talking to the printer via the serial line. Hence it should be a breeze to combine it with WirelessPrinting. @Colin66 @eben80 would you be interested to give it a try?

Hi guys,
I'm also working with very limited skills. My solution for communicating with the Serial port is extremely inefficient, probably the right way to do it would be what @luc-github uses in ESP3D.
My solution I currently have so it just about functions... my standards are very low :D

Some pointers:
I'm using SoftwareSerial for coms with the printer, the rxbuffer needs to be increased for lengthy responses like file listings.... on the hardware side I'm using voltage limiting zener diodes and a common ground connection to do the level shifting on both RX and TX.

probonopd, 2019-10-23:
Hi @eben80, may I be curious why you are using SoftwareSerial?

eben80, 2019-10-23:
> Hi @eben80, may I be curious why you are using SoftwareSerial?

@luc-github just asked me the same thing today :D
In my mind, I needed it because I wanted to keep UART0 for serial debugging...
I just realized that pin14 & 15 that I am using is connected to the SD Card. 
Since I haven't been using the SD Card, I haven't had any issues.

probonopd, 2019-10-23:
Well, in this project we need the SD card of course. But I think SoftwareSerial is not a must.

romandoss, 2020-04-12:
I run yesterday, but I had to solder wire for uart to 16 and 17 gpio directly to esp32.
And strange things, I have two esp32-cam, on one which branded AI Thinker after uploading sketch I get infinite bootloop, on other no name module worked fine.
And I also got worked sd card slot only after that modification "hasSD = SD.begin(13, SPI, fastSD ? 50000000 : 4000000);" I'm not sure correctly this or not.
And one more thing, if I reset 3d printer by button, serial gone to forever "#TIMEOUT#"

probonopd, 2020-04-12:
When you reset the 3d printer you also need to reset the ESP at the same time. Connect the reset lines of the printer and of the ESP together.

I don't know the answer to the other questions.

smartin015, 2023-04-16:
Not to muddy the waters too much, but there's now an ESP32S3 module with builtin camera: https://www.seeedstudio.com/XIAO-ESP32S3-Sense-p-5639.html

This board also appears to support USB host mode (crossover with #123). I splurged on a couple, should be arriving in a week or two. When I find the time to poke at them I can report back here with observations.

probonopd, 2023-04-16:
..."supporting SD card slot for external 32GB FAT memory" sounds good. Don't forget to order that, too.

smartin015, 2023-04-16:
> ..."supporting SD card slot for external 32GB FAT memory" sounds good. Don't forget to order that, too. We need to store the GCODE somewhere...

It's actually built into the module :D

probonopd, 2023-04-16:
I was mislead by the "external".

smartin015, 2023-05-12:
Small update - received the XIAO modules yesterday and took one for a test drive today with the [default code](https://github.com/limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/blob/main/CameraWebServer/CameraWebServer.ino). I could connect to wifi and stream video from the camera with zero issue. Pretty slick. I can also confirm it indeed comes with an SD card slot, although I haven't yet tried using it.

That's about as far as I'm going to test for now, until I'm finished with #123 on the S2 mini.