probonopd/WirelessPrinting #101: SD Card vs SPIFFS

ipepe, 2019-03-03:
I'm having issues with detecting and using SDCard shield. Every time I can see on my printer screen ip address and SPIFFS. I tried formatting sdcard on Mac, Windows,  with HPUSBFW.exe. I tried using 1GB, 8GB and 32GB but its still using SPIFFS. How can I debug that? Or did You remove feature of caching gcodes on sdcards? 

I'm running on 1129b07 with Wemos D1 Mini and MicroSDCard shield.

GMagician, 2019-03-03:
USE_FAST_SD? try to comment it and recompile

ipepe, 2019-03-03:
Uh. I'm still having issues with compiling it... I guess without recompiling, it won't work...

chepo92, 2019-03-03:
Did you try any SD example to make sure it works? Also update all your libraries

ipepe, 2019-03-04:
I'm not sure what the issue was, but one of my cards is working properly. It's 16GB Kingston Class 10 Card.

ipepe, 2019-03-04:
Actually I tried printing and it did not work. I managed to execute example code of https://github.com/greiman/SdFat/blob/master/examples/SoftwareSpi/SoftwareSpi.ino
but my pins are different:
```cpp
const uint8_t SOFT_MISO_PIN = 12;
const uint8_t SOFT_MOSI_PIN = 13;
const uint8_t SOFT_SCK_PIN  = 14;
//
// Chip select may be constant or RAM variable.
const uint8_t SD_CHIP_SELECT_PIN = 2;
```

Any idea how can I make this work with Your software?

GMagician, 2019-03-04:
I think you may change lib in your folder and recompile sketch

probonopd, 2019-03-04:
What kind of hardware are you using @ipepe?

ipepe, 2019-03-04:
https://wiki.wemos.cc/products:d1:d1_mini
https://wiki.wemos.cc/products:d1_mini_shields:micro_sd_card_shield

probonopd, 2019-03-04:
Now that is strange. I am using the exact same hardware (or a clone?) of it.

ipepe, 2019-03-04:
That's what I assumed. The weird part is that it reports on my printer screen ip adress and SD as filesystem. But apparently something is not working as it should be.

ipepe, 2019-03-04:
@probonopd Actually mine is probably some kind of weird clone. It does not have any Wemos logso or markings on it. And it looks like this: https://www.ebay.com/itm/New-Version-D1-Mini-NodeMCU-Lua-ESP8266-ESP12-WeMos-D1-Mini-WIFI-4M-Bytes-Module/254085828435

ipepe, 2019-03-04:
And MicroSD Shield is Lolin branded and marked as version 1.2.0

probonopd, 2019-03-04:
Pins 12, 13, 14 are not even there on the WEMOS D1 mini according to its pinout diagram.

ipepe, 2019-03-04:
https://wiki.wemos.cc/products:d1:d1_mini#pin
Look on left side and right side of table. Pin 14 is GPIO14 and that is marked as D5

probonopd, 2019-03-04:
I was looking at the "physical pins" in this diagram:
https://escapequotes.net/esp8266-wemos-d1-mini-pins-and-diagram/

What is a "port pin"?

ipepe, 2019-03-04:
```
The pins are what stick out of an IC, and connect electrically to the outside world. Ports are represented by registers inside the microcontroller, and allow the program (firmware) to control the state of the pins, or conversely, read the state of the pins if they are configured as inputs. There is a one-to-one correspondence between the pins on the microcontroller and the bits in its registers.
```
Source: https://electronics.stackexchange.com/questions/134605/pin-and-port-in-microcontroller

ipepe, 2019-03-05:
I'm when I'm trying to upload a file without "Print immediately" i get `ERR_CONNECTION_RESET` in Chrome. Could You confirm that uploading files through web ui works fine on master branch?

ipepe, 2019-03-05:
The problem grew even bigger. I successfully run examples:
**SdInfo** from SdFat
**SdFormatter** from SdFat
**listFiles** from SD (changed SS pin from 4, to 2 to make it work)
**SoftwareSPI** from SdFat (changed pins like is said before)

But failed with:
**SDWebServer** from ESP8266WebServer (SD Card not detected)
**WirelessPrinting** (0 file size after uploading, uploading breaks connection during process)

@probonopd Could You include Your file versions of:
`/Users/pptasinski/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/d1_mini/pins_arduino.h` and `/Users/pptasinski/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/variants/generic/common.h` ?
I had to modify my common.h file to change `#define PIN_SPI_SS   (15)` to `#define PIN_SPI_SS   (2)`

I could ask author of SdFat for help, but actually all of his examples work, so I assume there is some issue with Your code.

probonopd, 2019-03-06:
I don't think that these files are meant to be edited by users at all.

Have you tried by compiling the WirelessPrinting code yourself, or have you used the compiled firmware? (If you have compiled yourself, are you sure you have selected the D1 mini board in the Arduino IDE and not a generic ESP8266?)

ipepe, 2019-03-06:
I'm compiling myself. I'm sure I selected: `LOLIN(WEMOS) D1 R2 & mini`. 
I would assume that compiling examples from SdFat that work, would show some problem, but I don't have any issues when running examples code. Could You help me debug what's the problem with Your software? Any additional debugging code to help resolve this?

probonopd, 2019-03-07:
Where in our code do you think a change should be made? Sorry I am out of ideas since it is working for me and for others, and we don't even specify any pins for the SD card in our code.

ipepe, 2019-03-09:
I flashed aa1a5e5acb6efceef1e and seems to be working now.

probonopd, 2019-03-09:
Interesting. We have changed nothing regarding SD cards...

ipepe, 2019-03-09:
Maybe 0e3bec10099b4ccce989c211254face090733954 did some work? Maybe there was some race condition? Do You run Your ESP at 160MHz?

ipepe, 2019-03-09:
I think that we can close it for now

closed 2019-03-09 by ipepe