probonopd/WirelessPrinting #97: Discovering printer constantly

ipepe, 2019-03-03:
I updated to 10dc081 and ESP can't connect with printer. Im getting:
`Last received response: echo:Unknown command: "p�¯å¸‡˛Êò�ò‡òÊò�òÄˇM115"ok`

chepo92, 2019-03-03:
Same issue but the cause is other than the order, it has to try at least  times with the same baud

GMagician, 2019-03-03:
I checked code... and maybe with latest change I made two consecutive commands may be used again (that was not possible with first "big" change I've done to let everything works...I thinks that now we can go back adding two M112...

chepo92, 2019-03-03:
it was broken by #87

GMagician, 2019-03-03:
Yes! but first test of it was written with no possibility to send 2 messages and not receive one answer...final revision actually can let it work again so I think two M115 may be readded

chepo92, 2019-03-03:
> 
> 
> Yes! but first test of it was written with no possibility to send 2 messages and not receive one answer...final revision actually can let it work again so I think two M115 may be readded

I don't think that is the better solution, let me add some magic

probonopd, 2019-03-03:
@ipepe please try https://transfer.sh/10kbVS/ESP8266WirelessPrintAsync_d1_mini_413711c.bin and let us know if resolves this for you. Thanks!

chepo92, 2019-03-03:
This one should be working https://transfer.sh/a69dO/ESP8266WirelessPrintAsync_d1_mini_1129b07.bin !

chepo92, 2019-03-03:
@ipepe How did you connect your ender stock board with esp? as far as I know it doesn't have the serial pins to conect (I don't have an ender to try)

ipepe, 2019-03-03:
@chepo92 https://wiki.ipepe.pl/doku.php?id=ender3:esp3d_wirelessprinting

@probonopd will check in few hours

chepo92, 2019-03-03:
> @chepo92 https://wiki.ipepe.pl/doku.php?id=ender3:esp3d_wirelessprinting
> 
I imagined that was needed.  
So technicaly it's not stock, some hacking is needed

ipepe, 2019-03-03:
What's I meant by stock is that its not MKS. To be truly stock it should have stock firmware also.

probonopd, 2019-03-03:
We should attempt not to be too dependent on the particularities of any individual firmware, if possible.

chepo92, 2019-03-03:
> We should attempt not to be too dependent on the particularities of any individual firmware, if possible.

I don't see your point, marlin fw is like one of the standart (if not the most used), printer detection wasn't working for it so it wouldn't for any marlin-like fw

probonopd, 2019-03-03:
I agree.

ipepe, 2019-03-03:
I flashed 1129b07 and seems to be working. Although it is prone to freezing and slow at responding to requests.

closed 2019-03-03 by ipepe