probonopd/WirelessPrinting #105: [QUESTION] Web Firmware update

GMagician, 2019-03-08:
I'm wondering...no filename validation...no password protection...isn't it a little bit dangerous?

GMagician, 2019-03-09:
Another question..it's ok version on web page has no 2.0.x but only build number?

probonopd, 2019-03-09:
> I'm wondering... no filename validation... no password protection...isn't it a little bit dangerous?

For my own private use it's the most convenient, exactly what I personally need, no hassle. But I can see that not everyone might want this, so we could possibly use HTTP Basic Auth or something like that if a password is set for the OTA.

> Another question..it's ok version on web page has no 2.0.x but only build number?

Yes. I am not a huge fan of version numbers, I like git sha commit hashes. The commit hash should be directly linked to the commit, so that you can click on it to see exactly which code you are running.

closed 2019-03-09 by GMagician