probonopd/WirelessPrinting #109: Gcode Hooks

ipepe, 2019-03-09:
I think we could make some kind of "hooks" functionality. That would execute particular gcode files in `SDCARD/hooks/hook_name/file.gcode`. We could have hooks like: 
 * `SDCARD/hooks/after_print_finished/00_play_buzzer_finished.gcode`
 * `SDCARD/hooks/after_print_cancelled/00_move_up_and_left.gcode`
 * `SDCARD/hooks/before_print_started/00_something_something.gcode`

What do You think about that?

ipepe, 2019-03-09:
With that functionality we could potentially resolve theese: https://github.com/probonopd/WirelessPrinting/issues/91 https://github.com/probonopd/WirelessPrinting/issues/8

probonopd, 2019-03-09:
Personally, I don't like the idea that certain files need to be there on the SD card. Makes it too hard to use because the user would have to fiddle around with files on the SD card first.

ipepe, 2019-03-09:
We could create a button to create default hook files on sdcard. And with "fat" web UI sdcard file management could improve

probonopd, 2019-03-09:
Also, all functionality must work with SPIFFS only as well.

cpeuschel, 2020-03-16:
What is with a configuration file like marlin?

probonopd, 2020-03-16:
I am hesitant since it will increase overall complexity, both in code and in support. For me, the optimal software "just works" and does not need configuration.

cpeuschel, 2020-03-16:
We can add an page to upload custom GCODE into the SPIFFS?

probonopd, 2020-03-16:
Don't you think this sort of thing should better be done by the slicer software?

cpeuschel, 2020-03-16:
I am your opinion. Then we can close this issue, #91 and #8?

probonopd, 2020-03-17:
Closing for now since this can be done in the slicer, feel free if you think this should be reopened.

closed 2020-03-17 by probonopd