probonopd/WirelessPrinting #110: Better web ui

ipepe, 2019-03-09:
I offer my help to create better web ui. My skills are not spectacular regarding front end, but I think we could do better. I have few ideas that we could brainstorm:
 1. "Fat" but Pretty Web UI could be optional. It could work by sending file from `SDCARD/www/index.html` or if it's not present, we can render old html. If someone is working without SDCARD, we could provide "Fat" web ui as github pages with using API on local network device, for this we only need to set appropriate CORS headers.
 2. I quite like https://github.com/luc-github/ESP3D-WEBUI. We could "borrow" some stuff from there, especially moving motors with that pretty interface. Regarding controlling 3d printer a lot is possible because of sending GCODE commands by API.
 3. I thought about doing whole UI with Bootstrap+React. And contain everything in single html file with inline tags
 4. We could gather what people would want from such ui, like: "Printing progres bar", "Browsing SD Card files". A lot is possible in browser nowdays so I really would love to hear many ideas.
 5. I for purpose of development ui without ESP we could create simple NodeJS script that would "emulate" behaviour of ESP. This way, some frontend developers could easily join and help us, without knowledge and need for hardware or compilation in Arduino IDE.
 6. If we need to store something without ESP implementing API we could use localstorage.

Let me know what You think.

probonopd, 2019-03-09:
Is there a way to bake the HTML into the main firmware file, so that the user does not have to upload it to SPIFFS or SD? I like that by updating one firmware file, everything in the firmware gets updated. I don't like when the user has to fiddle around with additional files on SPIFFS and/or SD.

ipepe, 2019-03-09:
If the web UI will not grow big. We could embed it into CPP file.

probonopd, 2019-03-09:
Yes,
http://tinkerman.cat/embed-your-website-in-your-esp8266-firmware-image/

ipepe, 2019-03-09:
Wow. That looks really promising

probonopd, 2019-03-09:
Also check
https://github.com/ayushsharma82/ESP-DASH

probonopd, 2019-03-09:
> I quite like https://github.com/luc-github/ESP3D-WEBUI. We could "borrow" some stuff from there, especially moving motors with that pretty interface. Regarding controlling 3d printer a lot is possible because of sending GCODE commands by API.

Check how I send the "stop" command in the stop button. So, the basics are already there... especially since we have already implemented receiving random gcode commands for use with Cura.

ipepe, 2019-03-23:
So what solution would fit for You? How do You plan/imagine for this to work?

probonopd, 2019-03-23:
Having something similar to ESP3D-WEBUI which would send XHR requests to the existing OctoPrint-compatible enpoints might work, I guess

ipepe, 2019-03-23:
How about source code? Separate repository? Subdirectory in root of this repository? Stored in PROGMEM like in: http://tinkerman.cat/embed-your-website-in-your-esp8266-firmware-image/ ?

probonopd, 2019-03-23:
> Stored in PROGMEM like in: http://tinkerman.cat/embed-your-website-in-your-esp8266-firmware-image/

sounds good. This repository.

probonopd, 2019-04-10:
Check this out! https://github.com/ayushsharma82/ESP-DASH/pull/22

ayushsharma82, 2019-04-24:
Hi, Thanks for referencing ESP-DASH here. 
I think ESP-DASH would be great addition to this, I myself own a 3D Printer and i would love to tinker around adding a ESP8266 in it. Over that, ESP-DASH will provide data diagnostics and show temperatures on Web UI in realtime.

Recently, Slider Controls were added to the library and it can facilitate setting temperatures on the 3d printer via sliders. @malbrook is working on 4 new cards, Including Uploader Card which can be used to upload G-Code. ( https://github.com/ayushsharma82/ESP-DASH/releases/tag/v2.2.1 ).

probonopd, 2019-04-24:
Hi @ayushsharma82, welcome and thanks for joining here. 👍 

ESP-DASH looks like a great addition to this indeed, especially now that PROGMEM is being used instead of SPIFFS. 

Unfortunately I won't have the time to work on this myself anytime soon, so if you or @ipepe would like to go ahead and improve the web UI of WirelessPrinting you have my full support, I will definitely help with testing and will welcome pull requests.

ipepe, 2019-04-24:
I'm smashed with work recently so currently I have to say no. If anyone wants to step in, then go ahead.

ayushsharma82, 2019-04-24:
@probonopd @ipepe  No Problem. I will try to integrate whenever i get some free time.

probonopd, 2019-04-24:
### 👍 @ayushsharma82

probonopd, 2019-05-02:
@ayushsharma82 can ESP-DASH load values from JSON endpoints? We already have existing endpoints that e.g., Cura is using to show the temperature, print percentage, and such. It would be ideal if those could be re-used for the web ui.

ayushsharma82, 2019-05-02:
We can call `ESPDash.updateCard('id', value)` Inside those Endpoints.

probonopd, 2019-06-22:
@ayushsharma82 do you still think you can contribute something to get this started? Even a small starting point would be good (as an example), I then probably could do the rest.

ayushsharma82, 2019-06-23:
I'm currently on vacation. Will do it when I get back.

onfoot, 2020-09-16:
So, you moved to Mars permanently? 😆

ayushsharma82, 2020-10-20:
@onfoot  I'm swamped with work. Didn't get time to shuttle back to earth. 😆

Enjoy this view:
[image]

closed 2022-04-26 by probonopd