probonopd/WirelessPrinting #113: Testing with Wemos Mega

chepo92, 2019-04-12:
I found this [board ](https://es.aliexpress.com/store/product/WAVGAT-Mega2560-WiFi-R3-ATmega2560-ESP8266-32Mb-memory-USB-TTL-CH340G-Compatible-for-Arduino-Mega-NodeMCU/343255_32980303227.html?spm=a219c.search0104.3.9.458730e7s0422Z&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10547_319_10059_10884_317_10548_10887_10696_321_322_10084_453_454_10083_10103_10618_10307_10712_537_536_10713,searchweb201603_80,ppcSwitch_0&algo_expid=862dc920-7b5a-4bc7-9fb5-c6e9c68bf98d-1&algo_pvid=862dc920-7b5a-4bc7-9fb5-c6e9c68bf98d&transAbTest=ae803_4), has someone tested it with this firmware? I see it difficult to connect the sd card

probonopd, 2019-04-12:
This code is for ESP8266/ESP32; the board you linked has an ATmega2560. What would we do with it?

chepo92, 2019-04-12:
In my actual setup I have a ramps with a standard mega connected to a wemos D1 mini with a connector to the ramps, ideally this would have everything embebed (it is missing an sd card conected to esp)

probonopd, 2019-04-13:
I see what you are up to. I guess most users will only want to attach the ESP to their exsiting printer without changing the printer mainboard, though. If you want to do that, maybe Duet is a better option.

GMagician, 2019-04-13:
This board has a double CPU so probably you may use this code using SPIFFS without any issue nut I think that if SD pins are not shared between Mega and ESP8266 you will never be able to use SD. Some board schematic will probably help to decide what to do but I doubt you will never find them.

just-jason, 2019-04-13:
Some models seem to have a few pins from the ESP broken 
out, could you use a few of those for the SD ?

[image]

just-jason, 2019-04-13:
> This board has a double CPU so probably you may use this code using SPIFFS without any issue nut I think that if SD pins are not shared between Mega and ESP8266 you will never be able to use SD.

Is the current operation of the Wireless Printing code not exactly that? A "bridge" between the SD and the printer CPU through the ESP? There is no direct connection between the SD used for the Wireless Printing and the printer CPU. The SD is connected to the ESP ( looks like it should be doable in this case as well using the pins that are broken out) and the ESP is sending the commands through serial to the printer CPU ( in this case the MEGA) which is already set up in this boards configuration. I can not see why the SD card would need to be directly connected to the MEGA at all for use with Wireless Printing.

GMagician, 2019-04-13:
@just-jason that's my fault. I thought you meant to use SD connected do, eg, lcd board (that usually is connected to ramps). If you want to use an SD connected directly to ESP then you have to use some free pins (GPIO0 with care because is also used to enter program mode AFAIK).

just-jason, 2019-04-14:
> I thought he meant to use SD connected to, eg, lcd board (that usually is connected to ramps)

That seems to be the "holy grail" that everyone is looking for at the moment, but sadly not so easy to do in a usable way with usable transfer speeds. For now, this project is a good enough solution, and I thank you for all your contributions to it.

closed 2020-12-10 by chepo92