probonopd/WirelessPrinting #114: Connect printer using ESPUSB

balku, 2019-04-19:
HI, 
do you see a chance to use ESPUSB https://github.com/cnlohr/espusb to connect to a 3D printer instead of RX TX? 
I have a Creality Ender 3 Pro printer with usb and cannot find a possibility to use RX TX.
Regards
Kurt

probonopd, 2019-04-19:
If https://github.com/cnlohr/espusb/issues/45 ever gets implemented.

balku, 2019-04-20:
Many thanks for you quick answer.

closed 2019-04-20 by balku

just-jason, 2019-04-20:
> HI
> I have a Creality Ender 3 Pro printer with usb and cannot find a possibility to use RX TX.
 
Not one of the easiest boards to work with, but if you want to give it a try...
[image]

just-jason, 2019-04-21:
> If cnlohr/espusb#45 ever gets implemented.

Could this help?
https://github.com/tobozo/ESP32-3DPrinter-Bridge

probonopd, 2019-04-21:
Yes, that could work but at around $33 I think it's way too expensive; for that kind of money you could go with e.g., an Orange Pi Zero.