probonopd/WirelessPrinting #115: Convert to a Library

ayushsharma82, 2019-04-25:
I think, It would be best to convert this into a Library with callback functions and variables accessed like Temperature etc. It would provide a nice and easy interface for plugging in other libraries. More over, It keeps the clutter away from the user.

If we really want to expand this to common 3d printing users ( who have no experience in programming ), then it should not scare them at first.

Please let me know your views too.

probonopd, 2019-04-25:
Wouldn't that make it much more complicated for _us_? I have never really written a library.

ayushsharma82, 2019-04-25:
Nope. It would be a lot organised.  Just look at ESP-DASH on how to work with AsyncWebServer in a Class. 

The goal of a library is to provide a simpler interface for users while not creating a roadblock in developing of their own code. Some users might want to add some functions of their own when using this project, it's really hard to do that in a cluttered environment where the project creator only knows 'what goes where'.

probonopd, 2019-04-25:
It's certainly worse once not even the creator knows anymore 'what goes where'. I am hesitant since I have only basic Arduino language skills (can write sketches and consume libraries), but I am not really a C++ expert.

ayushsharma82, 2019-04-25:
It's really not that hard. 

```
class WirelessPrinting{
    public:

    private:
};

```

I will create a branch which is converted into a library. We will test this out incase it is better or not.

ayushsharma82, 2019-04-25:
So i have copy pasted the whole code into a Library ~ Class format. There are 4 ~ 5 errors on compilation, Any help would be appreciated.

https://github.com/ayushsharma82/WirelessPrinting/tree/library

GMagician, 2019-04-25:
I suggest to create a singleton class, I think that no more than one instance will be ever used

ayushsharma82, 2019-04-25:
@GMagician  I agree and i have done that by adopting Singleton Pattern 

Used `extern` .
https://github.com/ayushsharma82/WirelessPrinting/blob/library/src/WirelessPrinting.h#L131

GMagician, 2019-04-25:
@ayushsharma82 you should decvlarte all methods and fields as static...

ayushsharma82, 2019-04-25:
If possible, Can anybody make me as a collaborator?
 I can make a branch on this repo and all collaborators can contribute in completing it. I am not familiar with Singleton classes, @GMagician contribution will be highly appreciated.

ayushsharma82, 2019-04-25:
Or @GMagician  , You can clone my `library` branch in this repo.

GMagician, 2019-04-25:
> Can anybody make me as a collaborator?

I think is up to @probonopd add collaborators.

GMagician, 2019-04-25:
@ayushsharma82 you may also add a pull request here, using your code, then we can work on it...

probonopd, 2019-04-25:
I am not sure whether WirelessPrinting, being an application ("sketch") lends itself to being a library.
A library is for use in other sketches. WirelessPrinting is not for use in other sketches, it is a sketch itself.

So before we make things more complicated (to the point that I don't understand it anymore - what is a singleton class?) let's really think about what the pros and cons are.

GMagician, 2019-04-25:
Singleton class is a class with just one instance and usually without public constructor. Its instance is selfcreated by a method that return always the same instance.
Don't know real pros/cons , I just gave my opinion about how to do it

ayushsharma82, 2019-04-26:
It will just help us in future to have a clean code base to work with. You have all functions in `WirelessPrinting.h` header file to get a quick overview, and `WirelessPrinting.cpp` with all of them declared. If we don't organize this into a library then, It will be just time consuming for future development where we will have to read the whole .ino file for 'what we have done so far'.

Another Advantage is Distribution, Once we convert this to a library, We can use Arduino Library Manager to distribute to 'masses' (people who don't understand github / programming ).

ayushsharma82, 2019-04-26:
@probonopd  Can you create a library branch in this base repo? I will create a pull request to that 'library' branch. We can keep the branch open if it works otherwise we will just scrub the idea.

probonopd, 2019-04-26:
Done:
https://github.com/probonopd/WirelessPrinting/tree/library

ayushsharma82, 2019-04-28:
@probonopd  Done

ayushsharma82, 2019-05-13:
@probonopd I think we should stick with ino file only. Even i don't have time nowdays to completely overwrite the whole thing. 

Instead i will work on beautifying or sperating the current code like Marlin.

🙄

probonopd, 2019-05-13:
@ayushsharma82 please don't split things into may files, personally I find it much harder then. But your help on beautifying the web interface would definitely be most highly appreciated :-) Let's make it a top example for ESP-DASH.

ayushsharma82, 2019-05-13:
Roger.
I will only add esp-dash to it.

TheAssassin, 2019-06-12:
Might be useful to use the OctoPi protocol implementation on e.g., a Raspberry Pi Zero or cheap clones as a headless server with low overhead.

closed 2019-06-12 by probonopd