probonopd/WirelessPrinting #119: OctoPrint always busy

SSkull995, 2019-05-02:
Hi! 
I have a problem setting the wemos on my MKS 1.4. I flashed it with the firmware, no problems; I connected it to my wifi and I can connect to the IP on internet (the web GUI). I can also connect with Cura to it. However, as soon as I say "Print with OctoPrint", Cura says that "Octoprint is busy", thus no print can be done. Any suggestions? I connected it to AUX 1 on the MKS. 
Thanks in advance!

SSkull995, 2019-05-02:
Edit: I can't print from the web GUI as well

probonopd, 2019-05-02:
Which firmware and version are you running on the MKS?

Are you sure the ESP is communicating properly with the MKS?

Please connect to the ESP using telnet and observe the messages; please post the log here.

SSkull995, 2019-05-02:
Hi @probonopd, I actually believe that the wemos d1 doesn't talk at all. I connected to the aux 2 pins, like A11/D65 and A12/D66, but I never declared them in the .ino

probonopd, 2019-05-02:
A11/D65 and A12/D66 are MKS pins, right? You don't need to declare them in the WirelessPrinting `.ino`. You connect the WeMos TX/RX/GND/5V pins to the appropriate counterparts on the MKS, and that should be it.

SSkull995, 2019-05-02:
Yes they are the pins for the MKS. However, I don't know why, the printer is not accessible

SSkull995, 2019-05-02:
I just disconnected the wemos rx-tx and the result is the same, busy

GMagician, 2019-05-02:
You may check what do you get in ipaddr/info? and as already told you @probonopd check telnet
Note that wemos rx must be connected to MKS tx and wemos tx to mks rx...

GMagician, 2019-05-02:
Also...sure D65 and D66 are used as tx/rx?

SSkull995, 2019-05-03:
I will try the telnet later, sorry but yesterday I was a bit tired. I checked on the mks board and the tx and rx pins are located where I put the WeMos, very strange. I will check again in a few hours, as soon as I am back from the lab.

SSkull995, 2019-05-03:
Thanks a lot!

SSkull995, 2019-05-03:
Ok so, I checked the connection and...... TA DA! I found the problem, I connected the wemos to the wrong pins. Now it works like a charm, no more cables, except for Repetier Host

probonopd, 2019-05-03:
Glad it is working for you now.

closed 2019-05-03 by probonopd

SSkull995, 2019-05-04:
Hi @probonopd, I have another issue actually. When I send the code, the response is this:
{
  "files": {
    "local": {
      "name": "cached.gco",
      "origin": "local"
    }
  },
  "done": true
}

The printer doesn't even start

SSkull995, 2019-05-04:
From both Cura and the IP

probonopd, 2019-05-04:
Can you watch the output of the telnet port?

SSkull995, 2019-05-04:
I might have figured it out. I will let you know as soon as I check. What is the correct baud rate to upload the sketch? Arduino tells me that there is an error uploading

Kind regards,
Giovanni

SSkull995, 2019-05-04:
It says that espcomm open failed, maybe because it is connected to the main board?

Kind regards,
Giovanni

probonopd, 2019-05-04:
Where does it say "espcomm open failed"?

SSkull995, 2019-05-04:
Ok so I answer via email but nothing arrived. In the Arduino IDE, when I upload the sketch, in the bottom tab, it says uploading and then espcomm_failed

probonopd, 2019-05-04:
Do not upload to the ESP over USB while the ESP is attached to the printer.

probonopd, 2019-05-04:
You could use the OTA (over the air update) function, then you would not need the Arduino IDE nor USB...

SSkull995, 2019-05-04:
got it. I unplugged the esp from the printer and reduced the baud rate

SSkull995, 2019-05-04:
So now I am sending the code to wemos OTA through Cura, the bed is heating ahah. Will check once again. If I have to modify something, AFAIU I can do it OTA via the update right? What should I use as update?

probonopd, 2019-05-04:
Enter the IP address that is displayed on your printer's display after powering on into your web browser and go from there :-) you can upload the `.bin` file (e.g., from https://github.com/probonopd/WirelessPrinting/releases/tag/continuous) there.

probonopd, 2019-05-04:
Is it printing?

SSkull995, 2019-05-04:
On the Arduino IDE. Pic below

Kind regards,
Giovanni

SSkull995, 2019-05-04:
Ok it sent the message ahahah. It prints but when I cancel the print, I need to restart everything and clean the cache

SSkull995, 2019-05-04:
I believe it is normal

probonopd, 2019-05-04:
Yes. Seems like it is working for you now. What kind of printer are you using?

SSkull995, 2019-05-04:
Hi sorry! I replied via email but never arrived. I have a CreatBot modified, with MKS Gen 1.4 and some TMCs