probonopd/WirelessPrinting #122: List all the dependencies in README.md

ayushsharma82, 2019-05-13:
I think we should create a list of all required Dependencies with their specific version in README.md

probonopd, 2019-05-13:
Yes, please add to the ones already in https://github.com/probonopd/WirelessPrinting#building.

ayushsharma82, 2019-05-13:
I meant creating a markdown list.
( not everybody uses linux )

probonopd, 2019-05-13:
Hey, there is bash for macOS and Windows too ;-)
I think users can "translate". Let's not maintain the information in three places.

ayushsharma82, 2019-05-13:
You can consider readthedocs.io

They can be helpful.

probonopd, 2019-05-13:
You are right, but like everyone else I am short on time ;-/

ayushsharma82, 2019-05-13:
Understandable...

cpeuschel, 2020-03-07:
In my opinion, this can be closed, because we have now platformio and the dependencies is managed by platformio.ini

closed 2020-03-07 by probonopd