probonopd/WirelessPrinting #123: Work with printers without serial interface

TheAssassin, 2019-06-11:
Some printers' boards, such as for instance the Creality Ender 2, don't have RX/TX GPIOs for serial communication, which means this project cannot be used with them. As these are mostly inexpensive printers, using OctoPrint with more expensive hardware is annoying due to the price.

The printers however mostly have USB sockets which can be used for serial communication. Therefore the question is: can an ESP board communicate via USB with those?

My hope is there's a USB-to-serial sort of adapter that provides a USB host and translates the USB stuff into raw serial again. The cheap UART adapters you can find online are functionally equivalent to the stuff in the printer board, they lack the host feature and cannot be used.

I know some Mega2560 based Arduino boards can provide a USB host feature and might do the translation, however they're quite bulky and also expensive. It might be cheaper to get something like an Orange Pi or so and try to run OctoPrint on there (UI won't run very well, I hope there will be a completely headless version in the future).

Perhaps there's a kind of USB host chip to do the translation which doesn't require a complex circuit that needs to be soldered on a PCB, as most people don't have access to the tools to realize that.

probonopd, 2019-06-11:
Never say never...
https://github.com/cnlohr/espusb/issues/45

In the meantime, this is where you can solder to the Creality mainboard:
https://github.com/luc-github/ESP3D/wiki/Creality-CR-10-Ender-3

GMagician, 2019-06-11:
I tried to connect my esp8266 usb to mega usb (my printer).
On mega side I know it works because I usually use it with repetier
esp8266 TX and RX are connected to ch340 (or similar) and I thought, why not to use it and avoid soldering?   I failed...

TheAssassin, 2019-06-11:
@probonopd thanks, I might try later with those pins.

just-jason, 2019-06-12:
https://github.com/gdsports/esp8266-usb-host-demos

probonopd, 2019-06-12:
Actually there will  be a new ESP32 chip version with USB host mode soon: __ESP32-S2__. I think I am gonna support this one if possible, once boards for it are cheaply available.

Throwing in a USB host to serial chip is always an option of course, so this brings our options (at least) to:

- ESP32-S2 WITH USB (should become available soon)
- FT311D
- FT312D 
- CH341A
- MAX3421E
- Custom firmware on 24FJ64GB002
- Custom firmware on STM32F4

TheAssassin, 2019-06-12:
Ever tried either of these chips? I'm a bit sceptical, I couldn't find any quick examples for most of the chips that'd be equivalent to what shall be accomplished here.

probonopd, 2019-06-12:
> Ever tried either of these chips? 

No. But any chip that can convert serial-over-USB to serial should do, at least in theory.
In practice, most of this won't make sense from a cost perspective, compared to a Raspberry Pi Zero.

TheAssassin, 2019-06-12:
A Raspberry Pi Zero however doesn't run OctoPrint sufficiently smoothly...

probonopd, 2019-06-12:
I wonder if we can get this to compile on https://github.com/epsilonrt/piduino. Probably the WLAN stuff would need to be ifdef'd away...

TheAssassin, 2019-06-12:
> Who said OctoPrint? ;-)

Well...

> It might be cheaper to get something like an Orange Pi or so and try to run OctoPrint on there (UI won't run very well, I hope there will be a completely headless version in the future).
>
> https://github.com/probonopd/WirelessPrinting/issues/123#issue-454434678

Did I mention one of these cheap RasPi clones **melted** an SD card and it took me an hour to figure out what was wrong?

OctoPrint is great, if you have the right hardware to make use of all the features. However, if you can't, a small, energy-saving solution like WirelessPrinting is preferable.

probonopd, 2019-06-12:
> a small, energy-saving solution like WirelessPrinting is preferable

Agree! I was just wondering whether we can make it run on the [5 EUR Raspberry Pi Zero](https://www.berrybase.de/raspberry-pi-zero-v1.3) which has USB host mode...

TheAssassin, 2019-06-12:
Maybe by making this app into a proper C++ library and then building two frontends, one for Arduino, one for Linux hosts. That's the easiest and most manageable way.

TheAssassin, 2019-06-12:
A USB host solution by the way is also great for debugging, you don't have to solder to demonstrate this on arbitrary printers -> great for conferences.

probonopd, 2019-06-12:
ESP32-S2 :+1:

probonopd, 2019-06-15:
> I wonder if we can get this to compile on https://github.com/epsilonrt/piduino. Probably the WLAN stuff would need to be ifdef'd away...

Blocked by https://github.com/epsilonrt/piduino/issues/11 as it currently does not seem to support networking at all.

probonopd, 2019-08-04:
Looks like the [Geeetech 3D WiFi Module](https://www.geeetech.com/geeetech-3d-wifi-module-for-3d-printer-p-1018.html) can do it.

Looking at http://www.geeetech.com/firmware/Upgrade3DWF/config.txt the hardware is based on ESP8266 but the firmware is proprietary.

I wonder what chip/electronics they are using for the USB host. Someone who has the device should open it up and see what chip they have connected to the ESP8266.

probonopd, 2019-08-04:
Digging though their firmware gives `Cannot Find the interface for Communication Interface Class.` which is a string that appears in https://github.com/stm32duino/Arduino_Core_STM32/blob/9b6938fdc50f6d48b8f6f48eff4b1e96092fe666/system/Middlewares/ST/STM32_USB_Host_Library/Class/CDC/Src/usbh_cdc.c#L163 so I would not be surprised if they had a STM32F105/7 or STM32F2xx with USB host mode attached to the ESP8266.

probonopd, 2019-08-04:
**ESP32-S2** seems still not to be available yet on eBay.

TheAssassin, 2020-02-20:
> ESP32-S2 

Has this board ever been released? I don't see any of those in the respective shops.

Regarding the combination of an STM32 and ESP8266, it's also a pretty cheap and very universal solution. Ideally, we could simply combine a bluepill board with a NodeMCU or WeMos D1 Mini. I think that such a solution would work out of the box, provided that the STM32 just bridges between the USB serial interface and its own UART pins.
Using an STM32 might even allow for utilizing Marlin's "SD card reader" feature without having to use the serial interface, which can be painfully slow for larger G-Codes.

probonopd, 2020-02-22:
Yes, unfortunately the Blue Pill is lacking the needed USB OTG/host functionality.

The ESP32-S2 had been announced some time ago, but as far as I know there are no board with it on the market for sale yet. And also no code for the Arduino framework to use it as a USB OTG/host device  yet.

TheAssassin, 2020-02-25:
I've found some USB host shields for ~8USD, that still exceeds a RasPi Zero's price.

Perhaps this repository should be split up and modularized so that it's easier to build frontends for e.g., embedded devices, but also native ones? Some hardware abstraction wouldn't be too hard to implement, C++ makes it really easy (adding an abstract class which defines the serial communication interface as well as two or more implementations for the ESP8266/ESP32, Linux, and in the future perhaps even a simple Arduino Nano (should work for LAN, see #125)). Another advantage is that you can develop on your computer to debug non-hardware-related problems, e.g., the communication protocols. Unit testing also becomes a lot easier.

probonopd, 2020-02-26:
...or we just wait a bit more, then we will get Linux capable modules for the price of an ESP8266. It's only a question of time :-)

TheAssassin, 2020-02-26:
Linux is not a real-time system. But I guess that doesn't really matter here, since we just forward commands to an actual real-time system.

probonopd, 2020-02-27:
Yes. Actually, we can send a few lines "in advance" to what is being printed; hence the exact timing should not matter too much.

TheAssassin, 2020-03-22:
I ordered an STM32F4 for evaluation, which could be combined with a WirelessPrinting device, serving as a bridge between a USB CDC device and UART.

probonopd, 2020-03-22:
Which one did you order? Seems like STM32F401CCU6 boards are becoming cheaply available, but with that dreaded USB Type C plug (despite the chipset having USB 2.0 OTG FS).

Also note that not all printer boards implement USB CDC; some have chip-proprietary implementations like FTDI, Silabs, Prolific, WCH-IC CH340, etc., all of which you need host drivers for. Does STM32CubeMX contain all of those or "just" CDC?

probonopd, 2020-04-13:
ESP32-S2 support is beginning to trickle into the libraries... https://github.com/hathach/tinyusb/commit/2ff3f765db7e62201d0491093111b0d649446046

TheAssassin, 2020-06-18:
Someone mentioned the ESP32-S2? http://invidio.us/watch?v=L6IoSVdKwNM Looks like a nice device, let's hope for the best.

dlyckelid, 2022-02-02:
I am interested in developing a pcb with either ESP32-S2 or maybe a USB-Hostcontroller chip like MAX3421 and make it especially for this library with onboard sd-card reader or larger onboard volatile memory if that is better.

Has any tests been made with ESP32-S2 as USB-Host?

dlyckelid, 2022-02-02:
[image]
There you go!
Should be a good first prototype and if it works I can spend more time making sure it is more solid EMC and probably a bit better placing of the parts. But for test purposes this should work just fine :)

just-jason, 2022-02-02:
> [image] There you go! Should be a good first prototype and if it works I can spend more time making sure it is more solid EMC and probably a bit better placing of the parts. But for test purposes this should work just fine :)
> 
> The only way this makes sence is to buy a larger batch of them because otherwise they will be more expensive than a Raspberry Pi Zero 2 W that can run Octopie.
> 
> If I ordered 50 of them fully assembled the cost/piece is $7.4 + shipping and taxes
Do you have a schematic to look at?

dlyckelid, 2022-02-02:
[Schematic_ESP32 3D-Printer controller_2022-02-02.pdf](https://github.com/probonopd/WirelessPrinting/files/7989113/Schematic_ESP32.3D-Printer.controller_2022-02-02.pdf)

Sure here is the schematics.
I do need to look everything over again, Its basicly a collection of things Ive done in other projects copied and paste together so everything should work but I would put some more esd-protection, probably a cleaner powersupply and also a way to separate the 5v from the printer.
Also I need to check if there is any levelshifting needed for D+ and D-

But this was what I needed to get an estimate of size, and price

probonopd, 2022-02-02:
I have not attempted to do this with the ESP32-S2.

As for the MAX342, the code needed to support it would need to be written first...

dlyckelid, 2022-02-02:
Ok. I will do a little more tinkering with it and perhaps order some samples together with my next order of prototypes for other projects.
I have a number of creality-printers and well Raspberry Pi is to expensive (And I´ve also had stability issues with octoprint) and Pi zero 2 w seems impossible to buy

dlyckelid, 2022-02-02:
Would 16MB onboard memory be sufficent to skip the sd-card altogether?
The ESP32-S2 is available with 16MB onboard flash or you could use an external memorychip (Because the onboard flash is only safe for 10000rewrites).
I am using a W25Q128JVSIQ chip in another project and that is a pretty cheap useful chip. It is also non-volatile and 16MB and safe to 1000000 rewrites.

TheAssassin, 2022-02-02:
I don't think it's going to be sufficient. And non-volatile flash memory will be cheaper than those Winbond chips. Talking to microSD cards shouldn't be too hard, there's plenty of libraries for the Arduino platform.

I think that after all, writing a minimal, OctoPrint-compatible server which runs on some Linux host will be easier and cheaper.

probonopd, 2022-02-02:
> Would 16MB onboard memory be sufficent to skip the sd-card altogether?

I would say it highly depends on what models one wants to print. As for _me_, most of my G-Code files are actually much smaller than that. But there are notable exceptions.

TheAssassin, 2022-02-02:
https://www.winbond.com/hq/product/code-storage-flash-memory/qspinand-flash/?__locale=en might be a better choice, each chip contains 1 Gb of NAND flash (128 MB). Perhaps one could add as many chips as needed?

dlyckelid, 2022-02-02:
Just for starting to think about it I created this miniboard with onboard memory.
I will look into the NAND Flash if that is possible to include in this.
The board could be done both in micro and mini usb (Because I have printers with both)

But I am quite pleased with this. Would be really easy for anyone wanting their printer to be wireless :)
But ofcourse limited to 16Mb with this solution

[image]
[image]

Price for this one is completely different. Again if you just order 5 of them it will be about $10 for each depending on what price you get on the ESP32-S2.
But if you order in larger quantities like 50 you could get as low as $6/piece.

The reason I am looking at this is that I run a small electronics business and look for projects and be a little like adafruit that provide everything opensource for them who like to do it themselves but also provide finished boards for the bigger crowd.
Its just a small scale side hustle right now :)

TheAssassin, 2022-02-02:
For the record, I'd include the option to add some control buttons (e.g., by including pin headers). Most important button IMO is "abort print", that should trigger some "move print head away and turn off heaters" G-Code.

dlyckelid, 2022-02-03:
> 

That is a good idea. The first step is to ofcourse just see if it is possible at all to do it with the ESP32-S2.
Should be easy enough just to test if I can get the printouts from a printer to the esp.

I do have a small board with the USB-port connected to a ESP32-S2 somewhere so think I will start there just to test. The construction exercise was just to see if it is viable at all for me to spend my time on (More seriously) and because it was fun to attempt to do something as little as possible :)

But what I see could be possible when compared to other options is:
- Much lower price. With sd-card as memory a price sub $10 should be possible (Memory chips are expensive :( )
- Small footprint and no need for any extra cables, just plug it in and go
- Basicly no configuration, Just log in on AP, set the network SSID, password and start printing.
- More stable, My experience with octoprint and Rasberry Pi is not that good so I want something more reliable and simple

TheAssassin, 2022-02-13:
I just thought about your second proposal. I think you shouldn't add a specific connector (or do different boards for different types of USB connectors). The Ender-2, the printer I mentioned in the original post, wouldn't allow fitting a board right on the USB connector, which is exposed to the front through a little cutout in the acrylic cover. I always wanted to use a cable and mount the wireless interface in some place it fits into. Some printers might not even have space for such a board, so you'd have to mount it on the outside anyway. Your first prototype with the female USB type A looks most promising therefore. Just add a suitable (maybe even angled) USB A-to-[printer connector] cable and you're set.

dlyckelid, 2022-02-14:
> I just thought about your second proposal. I think you shouldn't add a specific connector (or do different boards for different types of USB connectors). The Ender-2, the printer I mentioned in the original post, wouldn't allow fitting a board right on the USB connector, which is exposed to the front through a little cutout in the acrylic cover. I always wanted to use a cable and mount the wireless interface in some place it fits into. Some printers might not even have space for such a board, so you'd have to mount it on the outside anyway. Your first prototype with the female USB type A looks most promising therefore. Just add a suitable (maybe even angled) USB A-to-[printer connector] cable and you're set.
> 
> Re. SD card vs. onboard flash: in the long term, SD cards will be the better option. You can just swap it for another one on failures, for example. Also, there's virtually no size limitation. The cards are really cheap, and I think many people have some old ones for such a purpose (I do have some 2-8 GiB ones I don't use, for instance, which could perfectly work as a cache for such a board). Also, with an SD card, some old g-code could even be kept for quick reprints (I'd like to have such a feature, also has some use in print farms for instance; of course that'd likely require a little display to be hooked up to the ESP).
> 
> I think the board should be discussed in a new repository. This issue is about adding support to @probonopd's firmware. Hardware questions are out of scope.

I agree totally with this. I think I was focusing to much on my own printers (Creality Ender 3 v2 and Creality CR-10s pro) Where the minimialistic design would work really well.
But there is other issues with this approach like manufacturing gets more complicated for hobbeist becuase of the small size and dual side components and so on.
I just ordered a pick and place machine and this will probably be one of the first projects I´ll do on it to test it out. I wil most likely go for something like the first design I made.

I will also create another repository for it when I get into that project more.

probonopd, 2023-04-16:
There are now very inexpensive ESP32-S2 based boards called the __S2 Mini__.

And Espressif seems to have a USB host mode driver for FTDI serial bridges and for CDC class compliant devices.

* https://components.espressif.com/components/espressif/usb_host_ftdi_vcp
* https://components.espressif.com/components/espressif/usb_host_cdc_acm

So possibly it could be done now.

smartin015, 2023-04-16:
Heya! That was a quick demo, more just to prove the concept than to be used in a "production" setup. I'm glad it caught your attention though, I'd be happy to help contribute here as this project is way further along. 

The examples I pull from all use the ESP-IDF framework. I expect building compatibility between the IDF stuff and your Arduino framework code would be step 0 here. WDYT?

probonopd, 2023-04-16:
Indeed, that would be tremendous. :+1:

smartin015, 2023-04-17:
Just a heads up that I started the port work today - successfully builds with VCP idf components included and imported into ESP8266WirelessPrintAsync.ino. I haven't yet tried using the VCP components or even running the firmware, but things are looking good so far.

WIP is in https://github.com/smartin015/WirelessPrinting/tree/esp32s2_usb_host.

It appears the [two recommended](https://community.platformio.org/t/using-esp-idf-library-within-the-arduino-framework-esp32/17226) ways of combining ESP-IDF and Arduino frameworks are:

1. Use `framework = espidf, arduino` which builds Arduino as an ESP-IDF component (this seems better supported, but adds a couple extra config files to the repo that are ignored when `framework = arduino`)
2.  Compile any ESP-IDF dependencies into static libs (`.a` files) and reference them from the arduino framework. This adds pre-build opaque dependencies into the project, but supposedly still allows just `framework = arduino` style operation.

I'm much more in favor of (1) as it's easier to debug compile-time errors and add more cross-platform stuff later on.

Additionally, I had to exclude the OTA library as I was running into the 1MB FLASH memory limit. Unsure if there's something less impactful I can cut down on, but I'm proceeding with that for now out of convenience.

probonopd, 2023-04-17:
> Just a heads up that I started the port work today - successfully builds with VCP idf components included and imported into ESP8266WirelessPrintAsync.ino.

Wow, that is awesome news! Very exciting. :+1

> I had to exclude the OTA library as I was running into the 1MB FLASH memory limit

Don't we have 4 MB flash on those lolin_s2_mini boards that we want to use for USB host mode?

We can do without the ElegantOTA library (in fact I wanted to remove it anyway), but we do need a way to update via the web interface (I have no physical access to the ESPs built into my 3D printers). Have a look at https://github.com/probonopd/WirelessPrinting/issues/175. We had web based updates working before we added the ElegantOTA library. The ElegantOTA library is purely cosmetical. Before the [PR that added the ElegantOTA library](https://github.com/probonopd/WirelessPrinting/pull/118), we had a simple (not so graphically nice but leaner) way of updating via the web interface, which hopefully doesn't need as much memory. Maybe you can add the original update mechanism back in, essentially reversing #118?

I hope we can keep SD card functionality for the lolin_s2_mini.

What is very important to me is that CI builds work.
Can you enable https://github.com/smartin015/WirelessPrinting/actions?

smartin015, 2023-04-21:
> Don't we have 4 MB flash on those lolin_s2_mini boards that we want to use for USB host mode?

Good catch, indeed we do - apparently the [default partition config](https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L5808) for the S2 mini reserves ~1M for the app, 1.5M for SPIFFS and the rest for OTA. I was able to minimize the SPIFFS and fit it all without disabling OTA.

> I hope we can keep SD card functionality for the lolin_s2_mini.

Shouldn't be a problem now :)

> What is very important to me is that CI builds work. Can you enable https://github.com/smartin015/WirelessPrinting/actions?

I'm still splicing things together - will enable CI once I'm ready to submit a PR. FWIW I've progressed to sending/receiving some USB serial data while the embedded server/telnet are up, but I'm hitting a snag with command interpretation (apparently my printer doesn't send an `ok` after handling an `M115` command). Dev work continues :)

smartin015, 2023-04-24:
Neat that you're making a custom board! And yeah, I had to power mine externally as well as I don't think there's anything in the USB spec about supplying any amount of voltage from the non-host device. Did you try my original code on it to confirm the VCP stuff is working?

I think there could still be a pretty elegant solution that turns the S2 into a little bump on a "USB cable", something like USB A to power the S2 on one end and USB micro to connect to the printer on the other end. One could buy and hack apart a standard USB micro cable to put it inline as well.

smartin015, 2023-04-24:
Another update on my end: code is still quite hacked together, but I've managed to get 

1. USB connection established and printer discovered (via M115 command)
2. Temperature auto-reporting parses successfully
3. I can run a gcode file from SPIFFS and it succeeds on the printer

**Video of all this working here: https://youtu.be/9qLVEFzK7UQ**

@probonopd before I actually submit my [draft PR](https://github.com/probonopd/WirelessPrinting/pull/181) and run the CI builds, I'd like your input on a few things:

1. I extracted several of the string parsing functions from `ESP8266WirelessPrintAsync.ino` and converted them from Arduino `String` to instead use `char*`; they now live in `lib/gcode/GCode.cpp`. This was due to the string corruption errors I'd mentioned before. I've also created unit tests for them at `test/test_gcode.cpp` which can be run via `pio test -e native -v`. Assuming I can get this to build for all platformio envs, is this a reasonable change to you?
2. Would you prefer the espidf components to be copied and checked into your repository (in the `components/` dir, adds a few thousand lines of "code") or would you prefer I look into using a git subrepository setup or a quick script to fetch and unpack them?
3. The `espidf, arduino` framework supports `log_i`, `log_e` etc. logging commands which have been very helpful for understanding what's going on. Would you prefer I include a couple of these to show the flow of gcode command send/ack and make them work for other boards, or strip them all out before I send the PR?
4. As a heads up, I've been having issues with compile-time errors relating to `'Serial' not declared in this scope`. I think this is some kind of config issue where the S2 only has `Serial1` and `Serial2`, but no `Serial` var defined. I've been working around this by commenting out some of the `Serial.println` lines in the libs used by this project, and I haven't actually found a real fix yet. If you have any ideas, I'd appreciate it :)

In other news, I'm pausing development for around ~2wks as I've got some other projects that need attention. [My PR](https://github.com/probonopd/WirelessPrinting/pull/181) should build and run just fine on an S2 mini however, so if anyone wants to pick it up and work on it they're more than welcome to keep going :)

probonopd, 2023-04-24:
> Unfortunately it does seem like atleast creality printers that I use does not provide a 5V source thru USB

Well, the printer is a device. The USB host needs to provide USB power. And in this case, this is the ESP.
But possibly you can get the 5V somewhere from the printer mainboard.

probonopd, 2023-04-24:
@smartin015 congratulations, I watched your video, this is awesome progress.

Regarding your questions:
1. Yes, let's do that. Please indicate clearly in the comments why it needs to be the way it is (or else you run the risk that next time  I touch the code I inadvertently fall back to old behaviors)
2. If we can have a separate `espidf` (sub)directory then I think it is easiest to copy those files (and their license file) there; I would not be surprised if sooner or later these functions would be integrated in a future version of the Arduino ESP32 core, at which time we should be able to delete this (sub)directly and change the include statements to use what the core provides
3. I'd say leave the debugging statements in, _if_ they don't cause issues while printing and uploading. Can the be enabled/disabled at compile time? (Bonus question, maybe for later: Would the debug statements make sense over telnet, too? At least in my case, I have no physical access to the ESPs which are built into my printers, so I use telnet to see what's going on.)
4. How about using `ifdef`s to check whether it is an esp32-s2 and then use the corresponding SerialX?

smartin015, 2023-05-12:
Okay - I should have time this week to make some additional progress here. I'll 

1. proceed with C string parsing (including comments)
2. relocate files to a new `espidf` directory. If using espidf with platformio requires them to be in specific locations, I can contribute a simple `setup.sh` to copy them to the appropriate locations so they're not cluttering the repo for non-idf users.
3. I'll add instructions to the readme for how to enable/disable the logs at compile time, and make sure there's no printing issues.
4. The errors happen in the supporting libraries that are downloaded by platformio, so I can't change them without maintaining a fork of each library. I'll keep searching for a solution.... I suspect it's related to modifying the console to output via UART.

probonopd, 2023-05-13:
> The errors happen in the supporting libraries that are downloaded by platformio, so I can't change them without maintaining a fork of each library. I'll keep searching for a solution.... I suspect it's related to modifying the console to output via UART.

Would it be possible to get the fixes into those libraries upstream? I think that would be the cleanest solution...

probonopd, 2024-06-30:
https://github.com/luc-github/esp32-usb-serial is now available, which adds support for

- CH34X
- CP210x
- FTDI FT23x

to ESP32S3 and (untested) ESP32S2.

Maybe someone would like to integrate it here?

smartin015, 2024-07-10:
Hey @probonopd - long time no update from me. Apologies for leaving things half finished, life got busy. 

I make no guarantees on additional work, but I did poke at the new arduino library you linked today. It builds with pio for esp32s2 (see https://github.com/smartin015/esp32-usb-serial-pio/) but I ended up submitting a PR to the original library as it was complaining about missing non-static members (see https://github.com/luc-github/esp32-usb-serial/pull/1).

Also had to specifically set the C++ version in `platformio.ini` as it was complaining about use of `auto` with C++11.

Haven't tested on a device yet; just wanted to leave some breadcrumbs in case another person wants a head start.

probonopd, 2024-07-11:
Thanks for the update @smartin015. I don't think anyone besides you is currently working on this topic, so in case you ever get it to work with WirelessPrinting, we'd certainly appreciate a pull request.

Elur12, 2024-10-01:
> Hey @probonopd - long time no update from me. Apologies for leaving things half finished, life got busy.
> 
> I make no guarantees on additional work, but I did poke at the new arduino library you linked today. It builds with pio for esp32s2 (see https://github.com/smartin015/esp32-usb-serial-pio/) but I ended up submitting a PR to the original library as it was complaining about missing non-static members (see [luc-github/esp32-usb-serial#1](https://github.com/luc-github/esp32-usb-serial/pull/1)).
> 
> Also had to specifically set the C++ version in `platformio.ini` as it was complaining about use of `auto` with C++11.
> 
> Haven't tested on a device yet; just wanted to leave some breadcrumbs in case another person wants a head start.

Hi, I've now looked at your code and augmented it a bit to work with the printer. The problem came out that the processing speed in Arduino-IDE is not enough, I haven't tested it with ESP-IDF (I don't know it).

Do you have any ideas how to speed up your code with Arduino-IDE?

probonopd, 2024-10-01:
Overclock the microcontroller maybe?

[2 more replies]