probonopd/WirelessPrinting #125: LAN version (no WiFi)

TheAssassin, 2019-06-12:
I'd like to see WirelessPrinting be able to interface via LAN with networks instead of focussing on WiFi only. That's a lot more secure (no need to burn keys (shared ones are especially annoying, never tried whether WPA2-Enterprise works with this software) and also easier to deploy at large scale.

Most Arduinos and many other chips can interface with specially made LAN adapters via serial interfaces. Shouldn't be too hard to adapt this software to work with one of those, should it?

probonopd, 2019-06-13:
Possibly something like https://esp8266hints.wordpress.com/2018/02/13/adding-an-ethernet-port-to-your-esp-revisited/ (W5500 Ethernet Network Modules) could be used. PRs welcome but I won't have the time to do it. I don't know how well it works with, e.g., Zeroconf which is crucial for me.

smartin015, 2023-04-16:
Just poking my head in on this issue as I'd bought some SPI ethernet modules specifically for doing something like this (part of https://github.com/smartin015/peerprint/issues/13). No guarantees, but I might be able to contribute something here once I've tested them out. I develop on the ESP32S2 but I do also have some D1 mini's I could try out for copmatibility.