probonopd/WirelessPrinting #129: SPIFFS and SD card file browser

probonopd, 2019-07-21:
SPIFFS and SD card file browser file browser so that we can store and manage more than one GCODE file on the SD card.

* For SD card, we could store all GCODEs ever sent to the printer as long as space permits
* For SPIFFS, we should only store GCODE which was specifically uploaded for this purpose (e.g., some simple calibration GCODE) in order to leave enough space

probonopd, 2019-07-21:
Possibly this can be de-prioritized if we have https://github.com/probonopd/WirelessPrinting/issues/131 working.

GMagician, 2019-07-21:
I've tryed some time ago to store a gcode into sd via repetier...unthinkable

probonopd, 2019-07-21:
This ticket is about storing on the ESP SD card, not on the printer SD card. #131 is about storing on the printer SD card.

GMagician, 2019-07-21:
SPIFF browser is already present (/editor) but not too much flash to store a lot of files

probonopd, 2019-07-21:
Yes, we need a way to print from there, and we need this for SD.

GMagician, 2019-07-21:
Print from /editor is almost impossible, it's linked in binary form I think and modify it is not as easy as it may seems