probonopd/WirelessPrinting #131: Upload to printer SD card

probonopd, 2019-07-21:
We could try to upload to printer SD card, although it is reported to be very slow.

probonopd, 2019-07-21:
* http://marlinfw.org/docs/gcode/M020.html M20 - List SD Card 
* http://marlinfw.org/docs/gcode/M021.html M21 - Init SD card
* http://marlinfw.org/docs/gcode/M022.html M22 - Release SD card 
* http://marlinfw.org/docs/gcode/M023.html M23 - Select SD file 
* http://marlinfw.org/docs/gcode/M028.html M28 - Start SD write
* http://marlinfw.org/docs/gcode/M029.html M29 - Stop SD write 
* http://marlinfw.org/docs/gcode/M024.html M24 - Start or Resume SD print

probonopd, 2019-07-21:
Most interestingly, we could use

* http://marlinfw.org/docs/gcode/M928.html M928 - Start SD Logging

Use this command to start logging all console and host input to an SD file while still operating the machine.

The way I understand it, we could print at normal speed and after a complete print would find the printed GCODE on the printer's SD card, from where it could be re-printed at any time.

Might also come in handy for debugging purposes.

GMagician, 2019-07-21:
It seems promising...

probonopd, 2019-07-21:
Trying it manually as we speak.

GMagician, 2019-07-21:
Perhaps I think printer still print sent code

probonopd, 2019-07-21:
I don't see the GCODE being saved, do you?

```
M22 ; Release SD card
M21 ; Init SD card
M928 log.gcode ; Start logging to SD card
; TEST
M29 ; Stop logging to SD card
```

telnet says:

```
>M21 
<echo:SD card ok#wait more#
<ok#ok#
>M928 log.gcode 
<echo:Now fresh file: log.gcode #wait more#
<open failed, File: log.gcode .#wait more#
<ok#ok#
>M29 
<ok#ok#
>M117 Complete
```

GMagician, 2019-07-21:
I can't test...long print... but

> <open failed, File: log.gcode .#wait more#

is not a good sign..

Loocking into marlin code M928 opens a file and set some flags, nothing more should be needed

probonopd, 2019-07-21:
With it, getting

```
>M21 
<echo:SD card ok#wait more#
<ok#ok#
>M23 log.gcode 
<echo:Now fresh file: log.gcode#wait more#
<open failed, File: log.gcode.#wait more#
<ok#ok#
>M928 log.gcode 
<echo:Now fresh file: log.gcode #wait more#
<open failed, File: log.gcode .#wait more#
<ok#ok#
>M29 
<ok#ok#
>M117 Complete
<ok#ok#
```

probonopd, 2019-07-21:
I am stuck here. Running stock Marlin bugfix-1.1.x firmware from https://github.com/probonopd/marlin-for-ender-2.

GMagician, 2019-07-21:
open in marlin calls:

>   if (!file.open(curDir, fname, O_CREAT | O_APPEND | O_WRITE | O_TRUNC))

and fail to this will send such error message.
I'm looking at marlin 2.0 source.

I don't like both `O_CREAT` and `O_APPEND` but I need to wait 4h before checking something

probonopd, 2019-07-21:
Tried with 2 SD cards, and also with `LOG.GCO` as the filename.

probonopd, 2019-07-21:
Prusa i3 MK2S says:

```
>M22 
<ok#ok#
>M21 
<echo:SD card ok#wait more#
<ok#ok#
>M23 LOG.GCO 
<ok#ok#
>M928 LOG.GCO 
<ok#ok#
>M29 
<ok#ok#
>M117 Complete
<ok#ok#
>M105
```

but I don't see the file on the printer's SD card either.

GMagician, 2019-07-21:
1h left then I'll have printer to test it

probonopd, 2019-07-21:
Having a transparent serial bridge #132 would come in neat to test things like this.

GMagician, 2019-07-21:
Tested now.. use file format 8.3

GMagician, 2019-07-21:
but as I said before, it execute commands while it save them into file

probonopd, 2019-07-21:
Yes, it would be good if it would print as usual and save a copy of what is being printed to SD. I could not make it work, though...

GMagician, 2019-07-21:
I just sent:
M928 test.gco
G28
G0 x0 y0
M29

GMagician, 2019-08-03:
That's may be a new approach, but only available on Marlin2 and must be enabled on firmware
https://github.com/MarlinFirmware/Marlin/pull/14817