probonopd/WirelessPrinting #132: Transparent serial bridge

probonopd, 2019-07-21:
Add a transparent serial bridge on port 2323 that would allow to interact with the printer as if it was connected to a serial port locally.

While a client is connected there, all other activity needs to be stopped, so that it does not interfere with the transparent serial bridge.

This should allow us to:
* Use printer host software like Pronterface and Atelier
* Flash the firmware of the printer using `avrdude`

probonopd, 2019-07-23:
https://github.com/probonopd/WirelessPrinting/pull/128 is insufficient because it does not stop all other activity while a client is connected.

probonopd, 2019-07-25:
Tried to implement this but getting

```
me@host:~$ avrdude -DV -p atmega1284p -c arduino -P net:192.168.0.33:2323 -b115200 -U flash:r:factory.hex:i
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
```

It seems that it understands that a connection is made and stops all other activity, but it does not seem to respond to the telnet client on port 2323 with what is coming from the printer's serial.

probonopd, 2019-07-25:
I managed to get the transparent serial bridge to work up to the point that I ca

```
telnet 192.168.0.33 2323
```

and interactively issue commands and get back responses. I was hoping that this would be sufficient for using `avrdude` to flash firmware to the printer, but unfortunately I am getting

```
me@host:~$ avrdude -DV -p atmega1284p -c arduino -P net:192.168.0.33:2323 -b115200 -U flash:r:factory.hex:i -c stk500v1 
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
```

Any help appreciated.

probonopd, 2019-07-25:
https://github.com/AlphaLima/ESP32-Serial-Bridge/blob/master/ESP32-Serial-Bridge.ino is using a buffer and is dealing with individual bytes - do we have to do this?

probonopd, 2019-07-25:
Maybe we need to use HTTP-based flashing instead as described on https://github.com/jeelabs/esp-link/blob/master/UC-FLASHING.md?

GMagician, 2019-07-26:
> Do we need to reset the printer mainboard using its RESET pin? (It is broken out on the ICSP header)

For mega2560 answer is yes, this is why Soc is reset when you activate usb connection.

probonopd, 2019-07-26:
OK, so I manually reset the printer by grounding the reset pin on the ICSP header in the moment when uploading starts. I get a little further, but only to run into more errors. So I think the timing of the transparent serial bridge may not be sufficient for flashing?

```
me@host:~$ avrdude -DV -p atmega1284p -P net:192.168.0.33:2323 -U flash:r:factory.hex:i -c stk500v1 -v -v -v -v -v 

avrdude: Version 6.3
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/me/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:192.168.0.33:2323
         Using Programmer              : stk500v1
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
             Memory Ops:
               Oeration     Inst Bit  Bit Type  Bitno  Value
               -----------  --------  --------  -----  -----
               READ               31     VALUE      7      1
                                  30     VALUE      6      0
                                  29     VALUE      5      1
                                  28     VALUE      4      0
                                  27     VALUE      3      0
                                  26     VALUE      2      0
                                  25     VALUE      1      0
                                  24     VALUE      0      0
                                  23     VALUE      7      0
                                  22     VALUE      6      0
                                  21    IGNORE      5      0
                                  20    IGNORE      4      0
                                  19   ADDRESS     11      0
                                  18   ADDRESS     10      0
                                  17   ADDRESS      9      0
                                  16   ADDRESS      8      0
                                  15   ADDRESS      7      0
                                  14   ADDRESS      6      0
                                  13   ADDRESS      5      0
                                  12   ADDRESS      4      0
                                  11   ADDRESS      3      0
                                  10   ADDRESS      2      0
                                   9   ADDRESS      1      0
                                   8   ADDRESS      0      0
                                   7    OUTPUT      7      0
                                   6    OUTPUT      6      0
[... 660 lines cut ...]
avrdude: Recv: . [03] 
avrdude: Recv: . [10] 
avrdude: Send: A [41] . [86]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [03] 
avrdude: Recv: . [10] 
avrdude: Send: A [41] . [87]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [89]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Vtarget         : 0.0 V
         Varef           : 0.3 V
         Oscillator      : 0.000 Hz
         SCK period      : 23925.8 us

avrdude: Send: A [41] . [81]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [82]   [20] 
avrdude: ser_recv(): read error: Connection reset by peer
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: B [42] . [82] . [00] . [00] . [01] . [01] . [01] . [01] . [03] . [ff] . [ff] . [ff] . [ff] . [01] . [00] . [10] . [00] . [00] . [02] . [00] . [00]   [20] 
```

GMagician, 2019-07-26:
> avrdude: Recv: . [14] 
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

That's a little bit weird...received 14 and expected 14..

probonopd, 2019-07-26:
Maybe received it too early or to late?

GMagician, 2019-07-26:
> avrdude: Send: A [41] . [80]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 
avrdude: Recv: . [14] 

This is a wrong answer. It should respond 14, value, 10

probonopd, 2019-07-26:
In any case, it's safe to say that my naive approach to it is not working. We'd probably need some magic along the lines of 

* https://github.com/jeelabs/esp-link/issues/50
* https://github.com/AlphaLima/ESP32-Serial-Bridge/blob/master/ESP32-Serial-Bridge.ino
* https://www.esp8266.com/viewtopic.php?f=6&t=864

but that is way over my head.

GMagician, 2019-07-26:
> jeelabs/esp-link#50

it seems like you get a 14 10 from previous query. because afer it you really get 14 3 10..Check if avrdude may increase timeout before resend command

probonopd, 2019-07-26:
> increase timeout before resend command

I don't think it can do it?

http://manpages.org/avrdude

CraigHoffmann, 2019-08-03:
Have you tried using this wifi bridge to control the 3d printer using something like repetier host?
Repetier allows you to select TCP/IP connection with ip address and port.

probonopd, 2019-08-03:
Yes, I had sent some codes and they worked.

ithinkido, 2019-08-05:
https://github.com/jeelabs/esp-link/issues/394

probonopd, 2019-08-06:
Thanks, so ESPLink seems to be able to do it. Does someone have the time to port it over?

ithinkido, 2019-11-03:
https://github.com/beckdac/ESP8266-transparent-bridge

probonopd, 2019-11-03:
Cool. How to best integrate into this firmware?

gmag11, 2023-01-23:
> Thanks, so ESPLink seems to be able to do it. Does someone have the time to port it over?

Hi. I'm using plain ESPLink in my 3D printer together with `socat` in my server where Octoprint runs. It works normally.