probonopd/WirelessPrinting #133: Wemos D1 mini v3

Kotik82, 2019-07-22:
Had previously a Wemos D1 mini pro with 16mb flash and everything was working fine (using the spiffs) but as i use that board for other projects i ordered a Wemos D1 mini v3 and a SD card shield for it.

After compiling and uploading to the new board i was able to connect and configure wifi but connecting it to the printer simply doesn't work, the device connects properly to the wifi network but it seems to be unable to communicate with the printer over UART. I checked and triple check my wiring and everything seems to be ok.

If i replace the D1 mini V3 with the D1 mini pro then everything is working fine again........

PS i tried the D1 mini V3 without the SD card shield, same results
PS2 runing Marlin 2.0 bugfix with #define BAUDRATE 250000

probonopd, 2019-07-23:
Did you compile yourself for both boards?

Kotik82, 2019-07-23:
Yes, compiled myself in both cases.

Kotik82, 2019-07-23:
Investigating some more, i see a small difference on the hardware side that might be affecting the issue.

The D1 mini V1 got a 470Ohm resistor on both the RX and TX pins while the D1 mini V3 got no resistors at all.

I need to find a 470Ohm resistor and solder it on the RX pin and see if that changes anything.

probonopd, 2019-07-23:
Strange, I am using a D1 mini myself, with a SD card shield. What happens if you use the precompiled one, with and without SD card?

probonopd, 2019-07-23:
Never seen a V3...

Kotik82, 2019-07-23:
Take a look here, i think it lists the Wemos mini models with their schematics, might come useful.

Kotik82, 2019-07-23:
Just uploaded the precompiled bin over OTA, same issue remains.

I don't have a 470Ohm resistor...... need to find one.

Kotik82, 2019-07-23:
Ordered the resistors but also got an idea, is it possible to define different pins for RX and TX, essentially redirecting them to different GPIO pins?

If yes then do you have an idea of how to do so?

probonopd, 2019-07-23:
I've done that for ESP32 but did not look into it for ESP8266.

Makes me wonder, does serial communication work at all on the V3 modules? Or do we have to consider them as defective?

Kotik82, 2019-07-23:
I got the resistors, will have to solder them in tomorrow and will let you know, there is always the possibility that i got a defective module, but while searching the web for similar issues i found another guy that has exactly the same issue with serial communication with a gps module and a V3 module.

Kotik82, 2019-07-24:
So, i soldered the resistors (470Ohm) on both TX and RX, still doesn't work. I am out of ideas, its either a defective board or Weomos D1 mini V3 has serial issues.

It is very strange cause everything else works, i can upload files i can upload OTA sketch but there is no communication with the printer.

I give up.

ipepe, 2019-07-26:
@Kotik82 Maybe try connecting Wemos to Wemos and check if serial is working?

probonopd, 2019-07-26:
Or connect the V3 RX and TX pins to a USB-to-Serial adapter and try talking to it both ways. We need to be sure that the hardware side of things is working at all.

Kotik82, 2019-07-26:
So i am back with more info.

Till now my theory was that since TX and RX pins of any ESP8266 are 3.3v tolerant (they might work with 5v logic but it is always better to run them with 3.3v logic)so i need to level shift the 5v TX of the SKR1.3 UART to 3.3v RX on the Wemos. 

Things i have tried till now:
1. 1k ohm resistor in series with RX and 2k ohm in parallel with GND ---> didnt work.
2. a zener diode (3.3v) in series with the RX ---> didnt work.
3. flashed ESP3D on the Wemos to see if the UART interface is working ----> working fine..........

So the UART of the Wemos D1 V3 is fine, there is something else going on.

Cause i can control the printer just fine with ESP3D on the same Wemos that refuses to connect with this sketch here.

ipepe, 2019-07-26:
Did You build yourself ESP3D or flashed downloaded binary?

Kotik82, 2019-07-26:
Build myself.

GMagician, 2019-07-26:
What about SKR TX level? if 5V maybe 3.3 is not good enough to drive it.
Also is GND common to both devices?

Kotik82, 2019-07-26:
and each time i build i make sure to erase whole flash with Wife settings inculded.

ipepe, 2019-07-26:
Can You also confirm that after flashing WirelessPrinting when You open built in Arduino IDE's Serial Emulator You can see M115 messages appearing?

Kotik82, 2019-07-26:
> 
> 
> What about SKR RX level? if 5V maybe 3.3 is not good enough to drive it.
> Also is GND common to both devices?

3.3v is usually fine on TX lines if u go from 3.3 to 5v logic on the TX side of things plus same hardware is working fine with a different sketch.

GND is common for both cause i am powering the Wemos fromthe AUX port of the SKR

Kotik82, 2019-07-26:
> 
> 
> Can You also confirm that after flashing WirelessPrinting on V3, when You open built in Arduino IDE's Serial Emulator You can see M115 messages appearing?

I can confirm that M115 does show up and under it i get garbage output till the M115 shows up again.

Kotik82, 2019-07-26:
But notice something interesting, the garbage output is not similar between the Wemos pro mini and the Wemos D1 mini V3
The pro mini output looks like this :

ph⸮ph⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115
ph⸮ph⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115

while on the mini v3 it looks a bit different:

hx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115
hx⸮`t⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115
`t⸮hx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115
hx⸮hx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮M115
M115

probonopd, 2019-07-26:
Are we sure 250000 baud are working properly? I have ever only tested it with 115200 so far.

Kotik82, 2019-07-26:
I reflashed Marlin 2.0 with 115200 yesterday and that didn't fix the issue. 

I also changed PrinterSerial.begin(serialBauds[serialBaudIndex]); to PrinterSerial.begin(115200); still no go... :(

Kotik82, 2019-07-26:
I am pretty sure there is some kind of voodoo involved :)

probonopd, 2019-07-26:
Just for fun, you have probably also tried Marlin 1.x? (I have never tested it with 2.0.)

ipepe, 2019-07-26:
You could also try on some really low baud like 9600

ipepe, 2019-07-26:
Or maybe it is some kind of simple hardware issue? Maybe try changing cables, or resoldier pins on Wemos? Maybe it's issue with voltage regulator on ESP? Try maybe powering it from usb charger?

Kotik82, 2019-07-26:
> 
> 
> Just for fun, you have probably also tried Marlin 1.x? (I have never tested it with 2.0.)

I didnt try 1.x, but your code is working fine on 2.0 cause i have been using it on the Pro Mini with no issues.

> 
> 
> You could also try on some really low baud like 9600.
> 
> Or maybe it is some kind of simple hardware issue? Maybe try changing cables, or resoldier TX/RX pins on Wemos? Maybe it's issue with voltage regulator on ESP? Try maybe powering it from usb charger?
> 
> You could still try connecting two Wemos devices together. You could upload a piece of code that flashed builtin LED when it receives M115 string or maybe any kind of string on to V3 and connect it to Your older Wemos that works.

Tried 9600 and while the LCD display is extremely slow with it the esp refuses to connect :(  i just now tried powering the esp over a power bank, same issue, also tried playing a bit with #define REPEAT_M115_TIMES 1, no change.

I triple checked my soldering and thats exactly what drives me mad, same Wemos device when flashed with ESP3D (same pins same cables same everything) works instantly. I am right know checking the ESP3D code, specifically the BAUD RATE part of it to see what they are doing differently.

It is not a big deal, after all a wemos mini v1 or v2 price is so low that i could order one and be done with it, i am just trying to figure out the issue out of curiosity and maybe save some headache in the future while learning something new :)

ipepe, 2019-07-26:
I'm interested also why this doesn't work when it should.

I edited my previous reply with:
> I also recommend that You should have single baud rate defined here: https://github.com/probonopd/WirelessPrinting/blob/master/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino#L64

probonopd, 2019-07-26:
> It is not a big deal, after all a wemos mini v1 or v2 price is so low that i could order one and be done with it, i am just trying to figure out the issue out of curiosity and maybe save some headache in the future while learning something new :)

Thanks for being persistent - this is interesting stuff indeed!

CraigHoffmann, 2019-08-03:
Hi, I had this issue with some Wemos D1Mini V3 clones from ebay.  The issue is that the TX pin from the CH340C is connected directly to the ESP8266 RX pin.  (the mini pro is a bit different)   When using the board by itself this is fine, but when you want to use an external serial device via the header pins it causes a problem as it seems in my case anyway the CH340C is never tri-stated so is always competing with the external TX you connect.  I haven't yet looked at the CH340C datasheet to see if it should tristate as a fixed it in the hardware.

As I have a electronic workbench always setup it was easiest just to cut the track after programming the Wemos.  of course the usb serial is then broken unless you resolder a connecting wire but that doesn't bother me.

I have attached some photos of where I cut if you wan to go this way at your own risk.

[image]
[image]

As was mentioned previously I think it would be much better to utilize the ESP8266 TX/RX pin swap feature and swap to different pins after power up.  This would also have the added benefit that all the ESP startup comms would not get sent to the 3D printer (this may be causing problems in an issue I see logged about flashing th 3D printer over the WiFi)

probonopd, 2019-08-03:
> As was mentioned previously I think it would be much better to utilize the ESP8266 TX/RX pin swap feature and swap to different pins after power up. 

I'd welcome a pull request that does it for ESP8266. I am already doing it for ESP32.

noega-33, 2024-11-10:
I had the same issue (not being able to use hardSerial connected to PZEM004) using Lolin wemos D1 mini V4.
I also found that V4 design was  different than original design (using CH340 chip)
in original design, ESP8266 chip RX and TX are connected to Board RX and TX outputs and  connected to TX and RX CH340 pins trough two 470 ohm resistors.
in V4 design ESP8266 chip RX pin is directly connected to  CH340 TX pin (no resistor) and ESP8266 chip TX pin is connected to a 470 ohm resistor before TX board output and then connected to CH340 RX pin
I resolved the issue by unsoldering and lifting RX and TX  CH340 pins and connecting 470 ohm resistors between pcb pads and lifted chip pins, as well as removing existing 470 ohm resistor ans replacing it by a small wire.
All work perfectly

noega-33, 2024-11-10:
1) Original Wemos design (that worked)
[image]
2) Wemos Lolin Wemos D1 mini V4 (HardSerial does not work)
[image]

My Solution: (see my previous post for description, below are some pictures)
[image]
[image]
[image]
[image]